slic3r/Slic3r #3406: Slic3r does not generate M109 commands if M109 is present in start G-code

lavato, 2016-07-10:
### Version
_1.2.9_

### Operating system type + version
_Windows 7_

### Behavior
I have a dual extruder setup and want to have a wipe script in my start g-gcode .

However when I put the wipe code in, Slic3r does not generate M109 commands at the beginning of the g-code file.
```
G28 ; home all axes
M109 S240
G29 ;  bed level
.....
```

This is not an issue when using a single Extuder but when a printer is equipped with two extruders and the user wants to print with the 2nd one or both, Slic3r will not put M109 in the g-code script. Therefore preventing the g-code script from executing.

alexrj, 2016-07-10:
That's intended behavior. I just realize the tooltip only mentions `M104` but it applies to `M109` as well:

[image]

If you're writing a custom heating procedure you have to write it entirely as Slic3r cannot reliaby guess what you want it to add and where. This is a feature.

The tooltip includes an example of how you can use the placeholder logic for integrating the commands in your start G-code.

Let us know if you need any help to write such G-code, or please close the issue.

lavato, 2016-07-10:
Thank you for your quick response. 
Could you explain what is the intended rational behind introducing this logic?

lordofhyphens, 2016-07-11:
The reason for the logic is because if you use M104/M109 in the start gcode, that indicates that you know what you are doing and Slic3r should not do extra things. Slic3r just isn't that smart to try to figure out what you did and I don't think it needs to be.

closed 2016-07-11 by lordofhyphens

mrvn, 2016-07-11:
One can write custom heating code but how does one make it heat the right extruder(s)? What if for this print only the second extruder is used?

What if the extruders have different temperatures? What will [first_layer_temperature] evaluate too?

lavato, 2016-07-11:
@lordofhyphens 

> Slic3r should not do extra things.

That just it, I think I am now clear on why this logic feels wrong, please see my arguments below:

* If I only set M109 T0 in my start code, Slic3r should set T1 temperature (or any other  extruder not specified in startciode). 
* If I set M109 T1 in my start code, Slic3r should set T0 temperature (or any other  extruder not specified in startciode). 
* If I set M109 T0 & T1 (or all extruders) in my start code, Slic3r should not set any temperature - same as what is happening now
* If I set M109  without specifying "Tx" in my start code, Slic3r should not set any temperature - same as what is happening now.

I guess I want to say is that Slic3r does not handle well multiple extruders  in this Start g-code logic and additional logic should be introduced to have the **same behavior** in multiple extruder as well as single extruder scenarios.

@alexrj @lordofhyphens @mrvn 
What do you think? Should this be raised for discussion?

mrvn, 2016-07-11:
Do you want per-extruder start code? Would that be before or after the start code? In the middle? where [extruder-start-code] is?

Should this be raised for discussion? I don't know. All I see is that the current way isn't optimal. As is I need different start code per print job depending on which extruders are used or not set the temp.

Thought: Maybe lines could be prefixed by the extruder and skiped if the extruder isn't used?

lavato, 2016-07-11:
@mrvn 

> Do you want per-extruder start code?

Hmm, I am not sure about a different start code section, but at lease I would like for Slic3r to acknowledge the fact that I am referring to a different extruder and set temperature for extruders not affected.

> All I see is that the current way isn't optimal.
Agree