slic3r/Slic3r #3407: Changes for Smoothie?

alexrj, 2016-07-10:
Discussion about this:
https://github.com/alexrj/Slic3r/commit/0673389009ff1b9fccaf765170afe6990aa363a1

lordofhyphens, 2016-07-10:
@wolfmanjm can describe what's going on; the additional G92 E0 required a nasty hack in Smoothie's new motion control code.

Even with the old firmware Smoothie treated E-only moves in units of mm/s (even in volumetric mode).

If you don't like it, get rid of it.

wolfmanjm, 2016-07-10:
smoothie now treats all E moves in volumetric mode as mm³ not mm, even retracts.

slic3r was inserting a G92 E0 after the G10 and before the G11, this casues G11 to leave E at the retract length. Newer slic3rs also put a G92 E0 after the G11 as well which is the correct place to put it.
Marlin has a nasty hack too to make G10 and G11 not actually change the E position at all which is very hacky IMO.

alexrj, 2016-07-10:
@wolfmanjm, in my opinion the semantics of G10 and G11 are not strongly designed and documented in both Marlin and Smoothieware. It's not clearly stated what the effect on E is after they are issued. In my opinion, the purpose of firmware retraction is to prevent messing up with the E axis at all, so I think they should *not* affect E. The issuer ignores the amount of expected retraction, so E gets into an unknown/unpredictable state after G10 is issued. I think G10/G11 should just perform the retraction/unretraction without affecting E.
Also, it's not clear/documented whether G10 and G11 are stateful or not (i.e. double retraction is prevented or not? I think it shouldn't.)
Moreover, I don't understand why you say that G92 E0 after G10 prevents G11 from working. Perhaps do you assume that E is always positive and can't go negative?

wolfmanjm, 2016-07-10:
@alexrj true G10/G11 are not specified well. However in the new smoothie it is not possible for me to do a G10 and G11 without affecting the E position. Hence the fix required.

There is code to prevent double G10 and G11 is only allowed if preceded by G10.

G92 E0 after G11 is good, G92 before G11 is bad.. in smoothie. It does not prevent G11 from working it does however leave the E at the wrong position after G11.

Nothing anywhere says that G10/G11 should not affect the E position, and in fact in the current implementation of smoothie this is not even possible, A move in E is a move in E and will be part of the motion control system and will register in the M114 position for E.

closed 2017-03-20 by alexrj