slic3r/Slic3r #3408: Printer Settings config is unexpectedly marked as modified upon loading

DrLex0, 2016-07-10:
### Version
1.30.0.25-prusa3d-release_candidate_1_3

### Operating system type + version
OS X 10.11.5

### Behavior
* When I load a Printer Settings config, often if will be instantly marked as ‘modified’. The modifications I have seen so far, are:
* Characters or character combinations in G-Code, like "--", "don't", "...", are replaced by single unicode characters: "—", "don’t", "…". This also happens while typing directly into the fields.
* ‘Only lift Z’ is sometimes considered changed. I haven't found a surefire way to reproduce this.

It seems like the G-Code fields are somehow configured for auto-correcting typographical fancy things. Of course this is _awful_ considering that the fields contain source code. So far, modifications have only occurred in comments, but if I would try to use one of those special character combinations in a status message to be printed with an M70 command, the printer would not be able to interpret the Unicode characters and display garbage instead.
And of course, the fact that the config seems to have changed while I didn't touch it, is annoying.

I haven't seen this problem in the 1.2.9 release.

#### STL/Config (.ZIP) where problem occurs
Try pasting the following in any of the G-Code fields of a config, saving it, then switching to another config, and back to the previous one.

; ------ Try pasting this G-Code ------
; Don't be surprised if it appears modified the next time you load the config...

alexrj, 2016-07-10:
I tried performing the steps you mentioned in the last paragraph, but I can't reproduce this behavior as of current master.

Also, prusa3d is a fork and we don't maintain it. So please report the issue to the fork maintainer or make sure you test with the original Slic3r repository.

bubnikv, 2016-07-10:
I am pretty sure that the same issue applies to the original slic3r. It is
likely caused by the wxWidgets 3.0.2 in a combination with some OSX
version. I will have to look into this issue.

Closed #3408 <https://github.com/alexrj/Slic3r/issues/3408>.

<https://github.com/alexrj/Slic3r/issues/3408#event-718001023>, or mute the
thread

bubnikv, 2016-07-10:
So this seems to be a general OSX / wxWidgets problem. Since OSX 10.9 the
smart dashes and smart quotes are on by default system wide. You may
disable them in system settings. There is currently no way to switch them
off through the wxWidgets api, but wxWidgets 3.1 disables that smart stuff
by default. Apart from the support of multisample anti aliasing, this seems
to be another reason to upgrade the wxWidgets to 3.1, but wxWidgets 3.1
break the perl wx module dependencies. I asked for help with wx module
support of wxWidgets 3.1 and got no response. I suppose the wxPerl has been
abandoned by the authors. Maybe a reasonable workaround would be to patch
wxWidgets 3.0.3 to force the smart dashes off, at least on our build server.
On Jul 10, 2016 11:22 PM, "bubnikv ." <bubnikv@gmail.com> wrote:

I am pretty sure that the same issue applies to the original slic3r. It is
likely caused by the wxWidgets 3.0.2 in a combination with some OSX
version. I will have to look into this issue.