slic3r/Slic3r #3412: Feature request: Take into account multiple extruder temperature definition in Start Code

lavato, 2016-07-11:
### Version
1.2.9

### Operating system type + version
Windows 7

### Behavior

**Current behavior - by design:**
When defining a custom script which involves setting temperature in the Start G-code section , Slic3r will not generate any M104 codes as it does not know what the user intends to do.

However in the multi extruder environment this behavior is not logical as Slic3r ignores the extruder definitions. E.g. M109 S240 T0

 **Proposed changes:**
If a user specifies an extruder for which they wish to write custom code, Slic3r should not generate any M104 codes for that extruder but should still generate it for others.

**E.g. In a setup with two extruders:**

* If there is M109 S240 T0 in the start code, Slic3r should set T1 temperature as it would if there was no custom code written.
* If there is M109 S240 T1 in the start code, Slic3r should set T0 temperature as it would if there was no custom code written.
* If there is M109 S240 T0  & M109 S230 T1 in the start code, Slic3r should not set any temperature - same as what is happening now.
* If there is only M109 S240 defined without specifying any extruder in the start code, Slic3r should not set any temperature - same as what is happening now.

I guess ultimately for a user, this behavior is more constant when in multi-extruder environment, as also per mrvn 's comments in issue #3406 .

Slic3r team, I hope the above makes sense. 
If not please feel free to just close this.

lordofhyphens, 2016-07-11:
I can see where it would be convenient for multiple extruders, and it looks like a logical/reasonable extension of the current behavior to me. 
So this change would semantically mean that for each extruder, if there is no M109/M104 code for that extruder then automatically generate code for it. 

There's a couple pitfalls with this though, which would either need to be accounted for or mentioned of "hey this does not work so don't even try it":
* The semantics of the T commands means that everything following it is treated as being for that tool, we'd need to consider this as we work through the start gcode. Not particularly difficult, but it is something that shouldt be considered. http://reprap.org/wiki/G-code#T:_Select_Tool
* Weird syntaxes for M104/M109. We'll need to write the parser to account for the current flavor selected. If someone tries to use one syntax which is incompatible with the current flavor, I think it should be ignored.

What say you, @alexrj?

lavato, 2016-07-11:
@lordofhyphens 

> So this change would semantically mean that for each extruder, if there is no M109/M104 code for that extruder then automatically generate code for it.

Exactly.

expertmm, 2017-06-09:
I came here when searching for my issue, but cannot understand whether my issue is the same or different. My mesh is set to "default" extruder (no change after importing). I try to print with left extruder of FlashForge Creator Pro, by going to Print Settings, Multiple Extruders, then changing all from 0 to 1. If I print, Slic3r uses the temperature from the first drop-down box under Filament in the main (Plater) tab, instead of from the second drop-down box.

I had hdglass in there so it was set to 220 where as the second drop-down was set to 240 PETG. When I printed, the LED screen on my 3D printer said target temp was 240. I cancelled the print, then changed the first drop-down also, so that both boxes were set to my hdglass setting. This workaround was successful. Here is my custom gcode used for "Printer Settings" (otherwise Slic3r outputs code which air prints as it does not raise the bed, which is a separate issue; note that I had to have a separate setting for left extruder, since markwal normally uses T0 for setting temp and homing during Start G-code--which is yet another issue):
Start G-code
```
; by markwal http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html
; make sure you get Slic3r version from that page not Cura version!
; begin
M73 P0 ; tell the firmware we're starting a build
G130 X118 Y118 A118 B118 ; set vref's to default
; let the Z stepper vref stay at eeprom level (probably 40)
;
; set temps
M140 S[first_layer_bed_temperature] ; set bed temp, don't wait
M104 S[first_layer_temperature_0] T1 ; set nozzle temp, don't wait
;
; home and set coordinates

T1 ; home on the left extruder
G90 ;  set positioning to absolute
G28 X Y Z ; home XYZ at default home feedrates
; home Z again slowly to give more consistent first layer height
G92 X0 Y0 Z0 A0 B0 ; define this as Z=0, the other coords are technically unnecessary but x3g requires them so gpx will make some up
G1 Z5 ; move the platform 5mm down
G28 Z F500 ; so we can home z again slowly
M132 X Y Z  ; Recall stored home offsets
;
; wait for heat up
G1 X110 Y-72 Z30 F3300 ; move to waiting position (front right corner of print bed)
M116 ; wait for target temperatures to be reached. GPX only, RepG doesn't understand
;
; purge and wipe
G92 E0 ; set current extruder position as 0 so that E15 below makes sense
G1 X110 Y-70 Z[first_layer_height] F2400.0 ; move to just on the bed
G1 X110 Y70 E15 F1200.000 ; extrude a line of filament along the right edge of the bed
G92 E0 ; set E to 0 again because the slicer's next extrusion is relative to this 0
```
End G-code:
```
;by markwal http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html
M73 P100 ; end build progress
M140 S0 ; set bed temperature to 0
M104 S0 T0 ; set extruder temperature to 0
M104 S0 T1 ; set other extruder temperature to 0
G1 Z150 F1100 ; send the build plate to the bottom
T0 ; Next job assumes T0 is the current tool
G28 X Y F2500 ; home X and Y axes
M18 ; disable all stepper motors
M70 P5 (We <3 Making Things!)
M72 P1 ; Play Ta-Da song
```