slic3r/Slic3r #3413: ExtUtils-CppGuess-0.11 test failure

Dinsmoor, 2016-07-12:
### Version
1.2.9-333-g5fb23f9

### Operating system type + version
Ubuntu 16.04 LTS 64bit

### Behavior
Configure failed for ./xs
Issue is with ExtUtils-CppGuess-0.11 failing to test.

Pull latest master.
Try to build IAW Linux/Ubuntu build instructions.

Expected to build/test.
Did not build/test.

#### STL/Config (.ZIP) where problem occurs

[build.txt](https://github.com/alexrj/Slic3r/files/358547/build.txt)

lordofhyphens, 2016-07-12:
This is an upstream issue apparently, nothing we can really do about it. 
As a workaround, set `export LDLOADLIBS=-lstdc++` before building. And once it's built you don't have to deal with it again. I'll add this information to the wiki.

closed 2016-07-12 by lordofhyphens

Dinsmoor, 2016-07-12:
If I install ExtUtils-CppGuess with ignoring tests via:

`sudo cpan -T ExtUtils::CppGuess`

I'm able to build Slic3r. with:
`sudo perl Build.PL && sudo perl Build.PL --gui`

and the application runs without problems.

I suppose this is just an issue with the tests.