slic3r/Slic3r #3418: Bridges generated in two sections.

ioquatix, 2016-07-13:
### Version
1.2.9

### Operating system type + version
Arch Linux latest.

### Behavior
Slicing bridges over enclosed area results in multiple areas which fail to connect.

See the attached .STL file and generated gcode.

ioquatix, 2016-07-13:
Previous layer:

[image]

Layer on top, it's not bridging correctly from outside to inside.

[image]

ioquatix, 2016-07-13:
Here are the STL, Gcode and Slic3r configuration.

[Issue 3418.tar.gz](https://github.com/alexrj/Slic3r/files/360767/Issue.3418.tar.gz)

lordofhyphens, 2016-07-13:
Is this  also present in 1.3.0-dev?

ioquatix, 2016-07-13:
@lordofhyphens I will see if I can test.. but to answer your question right now.. not sure.

lordofhyphens, 2016-07-13:
The gcode file you provided wasn't sliced with the same profile you included in the export bundle, btw. I don't know yet if that matters ^^;

lordofhyphens, 2016-07-13:
Also, what layers were in your screenshot?

Here's Z=5.15 sliced with your profile on 1.3.0-dev. It looks alright to my eyes. 
[image]

ioquatix, 2016-07-13:
@lordofhyphens that's weird, I sliced it and exported the config at the same time.. odd. Your screenshot looks good.. better than what I'm generating.. Just compiling 1.3.0 now

lordofhyphens, 2016-07-13:
The start gcode was different from my slice of it.

ioquatix, 2016-07-13:
I found that the issue did not show up in 1.3.0-dev so this has already been fixed. Thanks for your help and patience, and trying out the issue on your end.

closed 2016-07-13 by ioquatix