slic3r/Slic3r #3422: Perimeter inside bridge printed in free space

mistoll, 2016-07-13:
### Version 
1.2.9

### Operating system
Windows 7

### Behavior

When there's a hole in a bridge, the perimeter of the hole is printed without anything to print on being present. 

[image]
Left is ok, right does not work.

Simple object for reproduction
[image]

This is related to #657. 

#### STL/Config (.ZIP) where problem occurs
[config_stl.zip](https://github.com/alexrj/Slic3r/files/361637/config_stl.zip)

lordofhyphens, 2016-07-13:
Yep, that's an overhang and is basically impossible to print without
support.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

mistoll, 2016-07-13:
I agree that there are cases where there is no solution for this problem. The circle may be one of them.

Let's look a the rectangular hole. My idea would be to print a bridge aligned to one edge, omitting any bridge with a hole. After that print another bridge twisted by 90°, aligned to the other edge and attached to the bridge printed before. However I'm not sure if attaching one bridge to another would work.

You can see our workaround in the model. There we split the bridge into to layers.

FlameRunner, 2016-07-14:
If you insist on not using supports, try modelling with one-layer-height diaphragm. That kinda works even with circles.

mistoll, 2016-07-14:
That sound like a good workaround. I'll give it a try.

lordofhyphens, 2016-07-14:
Duplicate of #2621, with note by @FlameRunner as #254 (for Slic3r to do it for you).

Closing.

closed 2016-07-14 by lordofhyphens