slic3r/Slic3r #3423: Crash when using the "Send to printer" button

pjones, 2016-07-14:
### Version

1.2.9

### Operating system type + version

NixOS (Linux).

### Behavior

Click the "Send to printer" button.  Slic3r crashes and produces the following output:

```
Can't locate object method "new" via package "LWP::UserAgent" (perhaps you forgot to load "LWP::UserAgent"?) at /nix/store/yqg31pr2j9drjayy5id465bszhkr7x8z-slic3r-1.2.9/share/slic3r/lib/Slic3r/GUI/Plater.pm line 1230.
```

It would appear that the fix is to add `use LWP` at the top of the `Plater.pm` file.

lordofhyphens, 2016-07-14:
Please provide a copy of your Printer configuration (The complete exported config will work as well). This is important for testing purposes :) 
It's also helpful to know what you are sending to.

lordofhyphens, 2016-07-16:
Odd, I can't reproduce this on Debian Jessie with 1.3.0-dev, and it doesn't have any use LWP statements in Plater.pm. Since 1.2.9 is packaged with Cava I don't know of a good way to check if LWP is actually installed. There's an eval in GUI.pm to check for it.
Give https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-master--5fb23f9.zip a try, see if that works for you (use the .bat file)

lordofhyphens, 2016-07-18:
@pjones I'm going to leave this open for about a week; as I can't reproduce the symptoms I'm going to have to close it if you don't get back to me.

pjones, 2016-07-18:
I'll build a copy from source and see if that has any affect on the problem.

pjones, 2016-07-18:
I tracked this down to a missing dependency in the [Nixpkgs package](https://github.com/NixOS/nixpkgs/pull/17072).  I'll close the issue.  Thanks for the help!

closed 2016-07-18 by pjones

lordofhyphens, 2016-07-18:
No problem.