slic3r/Slic3r #3424: Use filament colling fan configuration with dual extruder

Martronic-SA, 2016-07-14:
### Version
1.2.9

### Operating system type + version
Windows 7

### Behavior
It seems that slic3r is not taking the second filament's cooling fan configuration when printing with dual extruder.
It's only taking the first extruder filament's cooling configuration and when switching to the second extruder, the fan remains at the same speed.
If you mix PLA and ABS it should take cooling confiugration into account when switching the exturder.

Don't take this into account if it has been fixed in latest version. I've tried to find a similar issue but haven't found anything.
Thanks!

lordofhyphens, 2016-07-14:
Try
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-master--5fb23f9.zip

That is a recent package of 1.3.0-dev. Use the .bat file.

lordofhyphens, 2016-07-16:
Doesn't look like it's been reported before with my search-fu.

Martronic-SA, 2016-08-18:
Maybe it too complicated to calculate cooling over layer time for some particular cases:
Most of the time, I'm not using min and max cooling fan speed, I set the same percentage for both and leave the fan at the same speed for the whole piece.
But I you print with a dual extruder for example PET (which needs cooling!) and IGUS (which dislikes cooling) I would like to turn off or at least decrease the cooling fan speed when the extruder switches to IGUS and vice versa.
Maybe it should be possible to set for each filament type the min and max fan speed to take.
For example I create a cooling model for the min and max layer time which keeps the increase and decrease cooling fan and for each filament type I set the min and max values.
So I could have 30%-60% for PET and 0%-5% for IGUS.

nicksears, 2016-08-22:
This is definitely a must-have feature.  It would be very nice for my research because I'm printing PLA and a UV cross-linking material (using fan output) so they need very different cooling profiles. But as Martronic-SA said, it's very important for printing any two materials with different cooling properties. 

I think the way it has to work is that when an extruder change happens, it changes to the logic for the filament for that extruder. It would calculate the fan speed slowdown, etc for the time it's printing. Could we get more info about what makes this complicated? Is it an issue for small islands or support material or soemthing to do with the order slicer does layers (extruder A layer 1, then B1 and B2, then A2 and A3, then B3 and B4, etc). I'd love to help and think this is a very important feature. 

Also, I'm looking to see if there's an open issue on this related issue now, but we really need to add support for multiple fan outputs. Then it should control each relevant fan while using each relevant extruder. This should happen whether we get this profile aspect fixed or not.

Martronic-SA, 2016-08-24:
In the same topic, That might be useful to allow the of control more than one cooling fan, maybe one cooling fan per extruder.