slic3r/Slic3r #3426: Repetier-firmware implements acceleration as M201/M202 seperately, not M204

lordofhyphens, 2016-07-16:
Repetier-firmware has a different gcode flavor than other Reprap firmwares (primarily Marlin). 
Definitely a case of https://xkcd.com/927/
Since https://github.com/repetier/Repetier-Firmware/issues/290 indicates that they aren't changing, we'll need to account for this hot mess.

lordofhyphens, 2016-07-18:
@alexrj this should probably be sorted out before the next release, as the current state of changing acceleration on Repetier FW is broken. I have a PR open for it but if you want to do something additional, say something along the lines of what you want and I'll add it to the PR.

lordofhyphens, 2016-07-20:
@repetier is there anything else that you know about that it handles differently than other firmwares that Slic3r might set? The attached PR covers acceleration.

repetier, 2016-07-26:
No I think only the acceleration stuff was changed some time ago in Marlin. M201/M202 date back to Sprinter I think. When marlin introduced M204 I was already using it for PID. Just be sure to use M201 which is for printing moves. I guess you never change travel acceleration.

- M201 - Set max acceleration in units/s^2 for print moves (M201 X1000 Y1000)
- M202 - Set max acceleration in units/s^2 for travel moves (M202 X1000 Y1000)

lordofhyphens, 2016-07-26:
The patch I have implemented currently sets both. I could probably do a
better job if I added a new field for travel moves. That doesn't do the
code cleanliness any favors though.

alexrj, 2016-12-17:
This was merged.

closed 2016-12-17 by alexrj