slic3r/Slic3r #3428: Feature Request: Extrusion Width for Bridging

timeshell, 2016-07-17:
Presumably the Extrusion Width for bridging should be equal to or less than the width of the hot end nozzle.  This would allow the bridge to be created without gaps and be self supporting.  The current Print Settings\Advanced\Extrusion Width doesn't give any option for bridge widths.

VanessaE, 2016-07-17:
Extrusion width only applies if there's something for the filament to squish down onto and hence spread sideways on.  Since bridging involves extruding in mid-air, there's nothing to make it spread out, so you only have a flow rate to consider.  This setting can be adjusted under Print Settings -> Advanced -> Bridge Flow Ratio.

timeshell, 2016-07-17:
I beg to differ.  The spacing expressed to the movement is important.  It may not squish, but the spacing between the lines needs to be very exact to minimize the horizontal gaps.  That is why it needs to be equal to or less than the nozzle width.

VanessaE, 2016-07-17:
That's not the same thing - spacing between lines is not necessarily tied to the width of those lines.  Look at, for example, support material.  You can set the pattern spacing in one section of the "Print Settings" profile, and in another section you can set its extrusion width.

timeshell, 2016-07-17:
The question isn't whether it "is not necessarily tied to width those lines".  The question is whether or not the bridging code is using the same extrusion width calculation as the rest of the infill or not.  Speculation doesn't help here.  What is the code actually doing?

VanessaE, 2016-07-17:
I can't speak for the actual action of the code, except that in my experience, the spacing of the bridge's perimeters is the same as the ends'/anchors' non-bridging parts, while the spacing of the first pass of infill lines varies with the flow ratio.

In any case, if you're extruding into mid-air, the extruded filament will always be the shape and diameter of your nozzle orifice, along with whatever die swell the filament exhibits.  About the only thing you could reasonably control here is the infill spacing, unless you want the code to draw the whole bridge separately from what's being laid down near the anchor points.

timeshell, 2016-07-17:
The shape of the extruded filament was never in question!  It is assumed it will be approximately the width of the nozzle.  What I am assuming is that the bridge print spacing is based on the extrusion width.  For example, if the calculated extrusion width is set to 150% of 0.4mm, 0.6mm width as the calculated width, then only 0.4mm of that width will be filled in the bridge leaving a 0.1mm gap on either side of each line printed, a total of 0.2mm when side by side with each line.

VanessaE, 2016-07-17:
You're correct as far as gaps between extruded lines and all, but you're still asking for a line spacing feature, not line width.  You can exploit the behavior of your filament's die swell via the flow ratio, but you still want a certain amount of plastic to be pulled across that bridge, so your control there is limited enough to basically be non-existent.

The way I see it, that leaves you three choices in practice:
* Lower the flow ratio and bump up the bridging speed, to take advantage of the filament's propensity to shrink/tighten as it cools, if any.
* Allow varying the infill spacing on the first bridge layer.
* Separate the bridge entirely from its anchor points and draw all lines therein with new spacing and flow ratio settings.

Only varying the infill spacing makes any sense, since the other two will undoubtedly lead to weak bridges.  It might be a useful feature.

timeshell, 2016-07-17:
Yes, line spacing feature is a great name for it.  I'm not good with names.  I'm good with technicalities.  Technically speaking, I'm still assuming that the "line spacing" at present is still being determined in code by the extrusion width.  I haven't looked at the code so I don't know for sure.  But seeing as there isn't any settings for it so far I can draw two conclusions.

1.  It already does what I'm looking for.
2. It is using the infill extrusion width calculation.  Which as such is the reason why I am describing it that way.

So, however you want to see it or call it, everything you see in this conversation will hopefully be enough to explain it.

VanessaE, 2016-07-17:
Right, the line width as it exists on the anchor points of the bridge is what determines the line spacing within the bridge, while the thickness of those lines is mostly the flow ratio.

Now mind you, the flow ratio does affect the infill line spacing, so while higher ratios will give wider line spacing, the filament will sag (unless you have really good cooling), and lower ratios will give finer spacing with equally finely-packed lines.  Meanwhile the perimeter spacing won't change, so you'll get more or less filament on those lines, if the filament stays in the bridge at all. :smile:

lordofhyphens, 2016-07-17:
Everything is flow to Slic3r, and you do have control over increasing/decreasing said flow for bridges. 

Could you start from the beginning and describe the problem at hand (preferably with pictures to show what is wrong from the output) before jumping to solutions?

timeshell, 2016-07-17:
Ok.  Basically, I want to eliminate gaps between the side by side lines while bridging.

Here is a scenario.  Create a bridge that is 4mm wide.  It needs to be filled with no gaps.  The infill ratio is 150%.   The nozzle is 0.4mm.  How many side by side lines will slic3r generate to cover the bridge?  Will there be gaps between the lines?  Is there any control over how it does this? Could these lines be made to overlap?

hyperair, 2016-07-17:
Just had a look at the code.

This is how bridge extrusion width is determined: https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/Flow.cpp#L81

For the C++-illiterate, it uses nozzle diameter as the width if bridge flow ratio == 1. Otherwise, it uses `nozzle diameter × sqrt(bridge_flow_ratio)`, so you can adjust your bridge extrusion width by setting this value to a value other than 1.

Furthermore, here is how the spacing is determined for adjacent bridge lines: https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/Flow.cpp#L44

For the C++-illiterate, it returns the bridge extrusion width + `BRIDGE_EXTRA_SPACING`, which is defined [in Flow.hpp](https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/Flow.hpp#L10)

I don't know why the extra spacing is added, but it's a pretty small value (0.05mm).

timeshell, 2016-07-17:
Ok, that's nice to know.  I think I'd still like the option to override that with a user set value though.

lordofhyphens, 2016-07-17:
I'm not entirely sure why it's a good idea myself, but perhaps @alexrj can shed some light on why it is the way it is.

bubnikv, 2016-07-18:
This is a very interesting discussion indeed.

To summarize: The bridges are extruded with a constant flow, and the flow is used to calculate the spacing to achieve exactly zero gaps of the extrudates at the anchor points. This leads to gaps once the filament gets unsupported between the bridges.

The solution @timeshell is asking for, is to keep the flow as usual at the anchors and to increase the flow once the nozzle leaves the anchors, and reduce the flow once the nozzle reaches the other anchor. The problem is the rapid change of extrusion rate, which is difficult to achieve reliably. One could probably bump up the extrusion rate before the edge of the anchor is reached, and hope, that the nozzle will squish the excess of material into the open space.

timeshell, 2016-07-18:
OR, increase the flow at the anchor points for the whole bridge and just overlap the flow.  Ideally, I would think, the goal in some way shape or form would be to have each each extruded line in direct contact with the previous line all the way down.  This should result is less stringing and less failed lines, in theory.

bubnikv, 2016-07-18:
We will not know whether the bridges would even be drawn reliably, if the neighboring strings touch. Maybe a thread will break if it gets dragged by the neighbor. We will not know until someone tries it. Or maybe @alexrj already tried?

to-the-nth, 2016-08-21:
From what I am seeing, the bridge anchor points are determined by normal line width spacing; extrusion widths on the advanced tab.  Those flow rates are used to the edge of the underlying layer, at which point, a new air bridge extrusion segment is started.  The flow rate of bridge is determined by nozzle width and bridge flow rate ratio as described by @hyperair, and has nothing to do with the layer height.  Most people are running with 0.4 nozzles, so the expected nominal bridge extrusion diameter is 0.4.  This is less than default extrusion width settings, so we wouldn't expect to see the bridge strands touching under normal conditions.  Back to @timeshell's problem statement of a 4mm wide bridge with no gaps:   can't get there from here without some gaps because the anchor point distances will vary based on the advanced tab settings.  The alternative is to modulate the bridge flow rate to try to fill the gaps, but I'm not sure that's a good idea.  In my experience, you really want a fixed flow rate for the bridge given a certain nozzle size, which is the way Slic3r supposed to work today.

Slightly off topic - 1.2.9a is pretty much unusable because of filament overlap bugs and filament over stuffing, and I have had to revert to 1.2.7 which is somewhat better.  Can someone tell me if the over stuff issues have been corrected in the latest experimental builds?  If anyone is interested in feedback, I'd be happy to create test cases and open bug reports if we think the situation is under control and we're ready for some QA.  

I wrote a graphical gcode post processor to correct some of the common cases  of even the 1.2.7 over stuff.  The post processor was also used to modify bridging gcode for Mojopix's Digital Sundial and produces 100% reliable bridging.  I found I had to identify and inflate bridge segments similar to what Simplify 3D does by about 0.5mm.  The other magic was to add a time pause of about 250ms - G4 P250 before and after the bridge to ensure good anchor attach.  Another effect noted was that with a 0.1mm layer height, the flow rate had to suddenly increase when starting the bridging segment, and it was important to be moving very slowly or pause as the extruder crossed in to the unsupported bridge.  In the end, the 0.5mm inflation and pause was more or less sufficient to also compensate for this issue.  The pause was especially important when a new bridge segment was preceded with a retract and move to new position.

I think Slic3r would benefit from the addition of a programmable pause at the start and end of each bridge segment.  Are there any plans to add a bridge inflate feature similar to Simplify 3D?

lordofhyphens, 2016-08-21:
@to-the-nth regarding the mass overfill for gap fill, it's largely been fixed in 1.3.0-dev. There's a Windows build on bintray (https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-add-libglut-dll--566bbbd.zip), which is off of the Windows CI server. Slic3r choosing the wrong infill width is still a problem, but that is easily worked around by specifying a fixed width for infill. 

"other slicer Y has feature X" is not a reason to add a feature X to Slic3r. Need to be able to justify it on its own merits (what problem does it solve? why is it a problem? is there another, better, way to solve it?), and even then it's more likely to get in if you produce a decent implementation of feature X rather than waiting for someone else to do it for you.

_extrude() in xs/src/libslic3r/GCode.cpp is probably where you want to look to insert a pause into the gcode, around line 565. https://github.com/alexrj/Slic3r/wiki/Code:-Adding-new-option-to-GUI-CLI has a list of the areas you need to touch to add new configuration parameters to Slic3r. 

If you get it working (and acting nice), submit a pull request and I'll look at it and give my recommendations to @alexrj. It'd help though if you opened a github issue first regarding adding pauses with the writeup as to why such a change may be necessary.

to-the-nth, 2016-08-21:
@lordofhyphens - thank you for the informative reply.

Just to make sure I understand, 1.3.0-dev infill should  be good to go if fixed values are specified in the Advanced tab extrusion width.

Last slic3r pull, build and hacks I did was about 18 months ago.  Is it still possible to build on Win10?  or do I need to be on linux?

You have confirmed that the addition of an bridge inflate feature is not in the queue.  I would be interested in working on this.  I will open a github issue discussion thread for this.  I'll include the Pause in the inflate issue since it relates to the starting motion/flow profile of a bridge segment(s).

Agreed "other slicer Y has feature X"  is not a reason, but I felt a little dismayed to learn that "slicer Y" had implemented the general case of inflate, compared to my gcode post processing hack.

lordofhyphens, 2016-08-21:
See wiki for building on Windows. I did it with Windows 10. The build
server runs 8.1 :)

lordofhyphens, 2016-08-21:
https://github.com/alexrj/Slic3r/wiki

There is a TODO page, which should also be referenced in README.md.

lordofhyphens, 2016-08-22:
@to-the-nth https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-master--566bbbd.zip is a 1.3.0-dev build.

alexrj, 2016-12-18:
I admit I haven't read this thread fully. Will do.

But do we have a test case? What issue are we targeting?
@timeshell, can you provide examples of problematic output by current Slic3r?

bubnikv, 2017-02-02:
@to-the-nth 

> From what I am seeing, the bridge anchor points are determined by normal line width spacing; extrusion widths on the advanced tab.

No. The bridge anchor points are determined by the nozzle diameter reduced by the bridging rate adjustment, and with 0.05 fixed distance added.

> Those flow rates are used to the edge of the underlying layer, at which point, a new air bridge extrusion segment is started.

The same flow rate is used for the anchors and for the bridges, therefore the anchors will be over-extruded. For tiny layer heights the over-extrusion will be quite extreme. If you see a change in extrusion rate between the anchors and the bridges, it is only that the extruder cannot exert enough pressure at the anchors, there is simply not space for the amount of material calculated for the bridges. Either the filament will act as a spring and then the extruder will release excess material when leaving the anchor, or the filament will slip at the hobbled bolt.

>  The flow rate of bridge is determined by nozzle width and bridge flow rate ratio as described by @hyperair, and has nothing to do with the layer height.

Correct.

> Most people are running with 0.4 nozzles, so the expected nominal bridge extrusion diameter is 0.4. This is less than default extrusion width settings, so we wouldn't expect to see the bridge strands touching under normal conditions.

This is not quite correct. The bridge lines are spaced to nearly touch with a (geometric) gap of a hard coded 0.05mm. What happens in the real world depends on how well the printer is calibrated, but the threads will likely not touch due to their slight sagging.

> Back to @timeshell's problem statement of a 4mm wide bridge with no gaps: can't get there from here without some gaps because the anchor point distances will vary based on the advanced tab settings.

That is not correct. The spacing at the anchors is dictated by the spacing of the bridge threads, not vice versa.

> The alternative is to modulate the bridge flow rate to try to fill the gaps, but I'm not sure that's a good idea. 

One may want to reduce the flow at the anchors to reduce over-extrusion.

> In my experience, you really want a fixed flow rate for the bridge given a certain nozzle size, which is the way Slic3r supposed to work today.

This is even more important for bowden setups.

> I found I had to identify and inflate bridge segments similar to what Simplify 3D does by about 0.5mm. 

I don't understand what you mean. You extended the anchors 0.5mm into the object? This is currently hard-coded into Slic3r, the value EXTERNAL_INFILL_MARGIN is set to 3mm and it is used to inflate not only bridging surfaces, but also the top & bottom sufaces. I believe it is large enough to anchor the top surfaces into the sparse infill.

> The other magic was to add a time pause of about 250ms - G4 P250 before and after the bridge to ensure good anchor attach. Another effect noted was that with a 0.1mm layer height, the flow rate had [...]

The programmed flow is equal along the anchor and along the bridge. But there is not space for the bridging extrusion flow rate to squeeze at the anchors. I trust you the pauses help to control the nozzle pressure better.

> In the end, the 0.5mm inflation and pause was more or less sufficient to also compensate for this issue. The pause was especially important when a new bridge segment was preceded with a retract and move to new position.

Surely the pause after deretract helps to equalize the nozzle pressure in any case.

I am really eager to try to understand what you did. I will print the sundial myself to test Slic3r. I think there is some misunderstanding on your side of how Slic3r places the bridges. I don't quite understand, how you added the 0.5mm inflation. In S3D or in Slic3r? What could S3D that Slic3r cannot in regard to bridges?

to-the-nth, 2017-02-12:
@bubnikv - Hi.  The Sundial is an interesting test vehicle.  You can OpenSCAD it down to one digit for a test case.  I also used the attached STL.  I apologize for not working on the Slic3r code as I had intended; life has totally gotten in the way, but I am happy to share my experiments.  Forgive me for not responding to each of your points individually.  But let's talk. 

Slic3r is one of the few slicing programs I have found that will do single filament wide vertical walls with arbitrary start and stop points.  (So I am not talking about building single filament object shell.)  I have only seen Cura do thin walls where the minimum thickness is two perimeter widths; one out bound and one return.   The upshot of this is Slic3r will often print two pillars with unbroken perimeters, and then come back and insert bridging. The first part of this type of bridge is a single filament adjacent to a wall which is no different from the single wall bridge case of the Sundial.   

The 0.05 over distance you mention is interesting.  Is that adjacent bridge line spacing or starting point to wall?  For reliable single first filament bridge attach I was using upwards of 0.50 mm over (into the wall).  (I wrote a program that scans and post processes the GCODE).  And yes, there is a small extra bump created from the plastic overflow.  But it tends to form a nice fillet.
[image]

If you think about what happens at the start a bridge with a arbitrary start point, there may or may not be a retract, and then a rapid move,  and then an un-retract, and then start feed with motion for the new bridge.  There would have been a blob of plastic, say 0.2mm high and 0.6mm wide that was under the nozzle, left at the old location before the move.  That blob needs to be restored at the new location before be can start the new filament.   Can't just move to a new location adjacent to an existing wall, un-retract, and start printing into mid air and hope that we will re-attach, (pretty much what Slic3r does today).   I think you have to move over the wall to some degree and extrude enough to re-attach.    What happens next is tricky, because as you cross the wall threshold at the start of new filament bridge, you have to form the new blob under the extruder.  That's where I think pausing for a few hundred milliseconds is helpful.  Gives it a chance for form. There's a lot of different possible strategies like moving to the threshold of a new wall, extruding a new blob and starting the new flow and motion for the bridge.

I also found that the attach for subsequent layers above the bridges was not sufficient.  Was seeing thinning/gaps.  Again it was the rapid move to new location adjacent to wall and start move/extrude.  I increased the start points to be pushed into the wall, but not to the degree as the first bridge filament.  

I post processed the gcode to change the anchor points and bridge flow rates.  Two weeks worth of coding and not 100% reliable.  I also think that specifying the bridge flow rate as a function of layer height or normal flow rate makes no sense.  When bridges are being printed, the filaments are round, so other than the first filament of a bridge which may need extra controls for reliability, the best we can do is stack the round cross sections together based on adjacent spacing, and maybe assuming there will be some extra adjacent attach (is that the 0.05?)

So part of what I was proposing was to inflate the size of the detected bridge area by a user settable param (more than 0.05) and then programmable pause on on all bridge move/restarts.  This is less sophisticated that what I did might might achieve similar results.

I can share some other mods that really helped the Sundial print quality if you are interested.

I am still running 1.2.7 because some of the later releases had very serious over extruding/flow rate issues.  1.2.7 still had issues, but it wasn't as bad.  The post processing does some flow rate reduction/correction and has simple graphical rendering of where the issues are.   Correct me, but it seems that Slic3r does not use adaptive extrude widths, rather they are fixed according to the settings in the Advanced tab.  So sometimes there are large gaps between adjacent filaments.  I did play with post processing to add flow to fill some gaps, but it quickly gets ugly.  Need to both move end points and change flow rate.  Right way to fix this is to allow adaptive widths when the original STL area slices are available.  Still not a trivial problem.

[Test Objects.zip](https://github.com/alexrj/Slic3r/files/769534/Test.Objects.zip)