slic3r/Slic3r #3429: Can't call method "size" when reading stl exported by blender.

alejandro-anv, 2016-07-17:
### Version
1.2.9

### Operating system type + version
Linux 64 bits

### Behavior
Open Blender, set units to meters, resize the cube to 2cm x 2cm, export as stl with scale=10000.
Load the file with Slic3r. All looks ok. Then click preview: display is empty. Click layers: after clicking anywhere Slic3r aborts with "Can't call method "size" on an undefined value at /home/anv/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 179."

I'm attaching the file
[cubo1.zip](https://github.com/alexrj/Slic3r/files/367850/cubo1.zip)

lordofhyphens, 2016-07-17:
Can't reproduce with my config.ini on 1.3.0-dev.

[image]

Please include your config.ini so I can check with that, also please do a build from source and check on 1.3.0-dev.

alejandro-anv, 2016-07-17:
config.ini?

I'm attaching all my ~/.Slic3r. I think this may be what you need...

[dotSlic3r.zip](https://github.com/alexrj/Slic3r/files/368014/dotSlic3r.zip)

alejandro-anv, 2016-07-17:
I Just renamed .Slic3r directory, started again with a clean one and it worked OK. Then it may be something related to bad settings...

My problem is solved but I think it could help you to trap the error and avoid the program abort.

lordofhyphens, 2016-07-18:
config.ini is the default name for the exported config. When an issue is posted, we need an export of that config that you were working with. Your .Slic3r directory is not particularly helpful for most problems. With your directory all I see is a "Simple mode" with an invalid print/printer configuration (layer height > nozzle diameter). I still couldn't reproduce your error message on 1.3.0-dev.

[image]

As I can't reproduce in current master, closing.

closed 2016-07-18 by lordofhyphens