slic3r/Slic3r #3431: Volume calculation for model is -1.

lordofhyphens, 2016-07-18:
[image]
### Version
1.3.0-dev 848374602d777f0795a6345c1b337a0d09c0eb4a

### Operating system type + version
Linux Debian Jessie x64

### Behavior
Slic3r is reporting STL volumes as -1.00. See screenshot. 
[image]

#### STL/Config (.ZIP) where problem occurs
[test.zip](https://github.com/alexrj/Slic3r/files/368332/test.zip)

bubnikv, 2016-07-18:
This is interesting. At least the Prusa3D Windows build gives a positive number.

lordofhyphens, 2016-07-18:
Yeah, I suspect it was something recent; I'll check on my build on bintray
and see if that's got the same bug in it.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2016-07-20:
Fixed, thank you.

closed 2016-07-20 by alexrj