slic3r/Slic3r #3433: Best Practice: Don't Throw Strings as Exceptions

lordofhyphens, 2016-07-19:
The methods in the ConfigBase class all throw strings (from what I can tell, C-strings at that) as exceptions. Unfortunately, this gets passed up the chain as an "unknown" exception which Perl doesn't know what to deal with. This is not a particularly good practice IMHO and made trying to fix a bug that showed up when the load method was ported to XS irritating to solve (error messages printed were useless and Slic3r was hard-crashing).

As an alternative, I recommend picking one of the standard exception classes such as std::runtime_error or something related to it in Boost. Since they'd be replacing strings, std::runtime_error is very likely sufficient for our needs.

lordofhyphens, 2016-07-19:
Looking over the rest of the codebase, I'm seeing that Model and Config are throwing static strings still. 
Other exceptions used on the C++ side:
* Clipper (which afaik is an upstream project) has its own exception type, which inherits from std::exception directly.
* poly2tri (again, pretty sure that's upstream) throws std::runtime_error. 
* GCodeSender is throwing a boost system error. 
* Print is throwing a custom exception, PrintValidationException (which inherits off of std::runtime_error).

I think that we could reasonably add a new exception class, derived from std::runtime_error to cover invalid options. Otherwise converting those to be std::runtime_error would also serve our current needs. 

The uses in Model.cpp could also easily be std::runtime_error. I don't think it's necessary to create a class for those; may be a target for further refactoring.

alexrj, 2017-03-17:
I agree with you.

commit 9f0335b4 (2017-06-04): Replaced throwing strings with exception objects. Fixes #3433
closed 2017-06-04 by lordofhyphens