slic3r/Slic3r #3436: Infill created inside a hollow object

Tinchus2009, 2016-07-20:
### Version
1.3.0 DEV - eb491056d88876cc9bd6e958191325ec499cb04a

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
win10

### Behavior
I tried to print a cooler. I have already printed it in the past. This time I have tried many configurations but for some reason, infill layers are created inside the object, wich is hollow. In other slicer softwares, the object slices correctly, and in slic3r 1.2.9 I remembered it sliced correctly too
Atached is the STL file and the config file

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

[ventilador.zip](https://github.com/alexrj/Slic3r/files/374719/ventilador.zip)
[config.zip](https://github.com/alexrj/Slic3r/files/374720/config.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-07-20:
1.2.9, Windows 7:
[image]

It's still doing it in 1.2.9. That STL is being parsed by the split function as many parts, which might have something to do with it. I don't know; the model is showing as manifold. 

1.1.7 might be worth taking a look at as well. You'd need to use a separate gcode viewer to get (approximate) pictures of the toolpaths.

Tinchus2009, 2016-07-21:
Your screencap is exactly what Im seeing. If you used 1.2.9 and got that results then is clear I was wrong. I cant remember then wich version I ised to print that object. Im sure I printed, I was useing it till it broke...

lordofhyphens, 2016-07-21:
Yeah, I didn't think there was anything actually wrong with the object, just that slic3r may be parsing it in a weird way is all.

mrvn, 2016-07-21:
git bisect is pretty efficient
ccache helps a bit too

Tinchus2009, 2016-07-21:
@lordofhyphens , I will. So far, last night I tested the following: If in DEV version I increase layer height to 0.3, the object is sliced correctly, the problem looks like when I use a 0.2 layer. The same happens useing previous version of slic3r.
I also tested version 1.1.7 stable, the same problem: it works useing 0.3 layer height, but the problem appears with a 0.2 layer, but this time, there is no infill/bottom layer (I actually believe that what is being created is a bottom layer), insead, there are missim layers on the object (atached is a screenshot) I will test other version tonight and update.
[image]

lordofhyphens, 2016-07-21:
Thanks for running this down more. At least we know a condition where it
works. If it was doing this in 1.1.7 then odds are the problem has always
been with us.

Tinchus2009, 2016-07-21:
Ok, I went back till version 0.9.7 The problem there still exist but: the object slices almost perfect. Only in layer n40 there is a solid infill created and then the rest is perfect, all other version tested creates many of theese solid layers. And also in0.9.7  if layer height is set to 0.3, the problem desapears.

lordofhyphens, 2016-07-21:
Wow, thank you for being this thorough.