slic3r/Slic3r #3440: Feature request: Draft shield support

alexxy, 2016-07-25:
It will be cool to have feature like a 'Draft shield' in cura. (currently its possible to set skirt height to needed value, but if it can be done in a automatic way...). This feature allow to print large objects in ABS without thermocamera...

https://ultimaker.com/en/resources/20426-experimental

lordofhyphens, 2016-07-25:
So what you want is the ability to specify "skirt height equal to object height"?

alexxy, 2016-07-25:
Right. I can do it manualy. But automatic way may be bettter =)

lordofhyphens, 2016-07-27:
Set the skirt height to -1. Not very intuitive, unfortunately. I'll look into implementing this as a UI checkbox.

Patola, 2017-03-12:
Not only that. A good warp shield / draft shield follows the vertical outline of the object closely (without the steep angles, of course). A good algorithm for generating that is described here: https://hal.inria.fr/hal-01141706/ (it is the algorithm used in the IceSL slicer).