slic3r/Slic3r #3441: Joining polygons if very close

ImInSpace, 2016-07-26:
### Version 1.2.9

### Operating system Win7 64 bits

### Behavior
Slice to svg seems to join together polygons if they are very close together.
Not sure if there is a setting to prevent this.

#### STL/Config (.ZIP) where problem occurs
[Bracelet.zip](https://github.com/alexrj/Slic3r/files/383809/Bracelet.zip)

lordofhyphens, 2016-07-26:
Could you try 1.3.0-dev? Doing a "from source" build is pretty straightforward (see wiki), but the following link should work as well. 
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-154#files

ImInSpace, 2016-07-27:
1.3.0-dev still has the same problem. I haven't been able to check it through the gui, but looking at the svg, the error is still there.

For referece, i used: 
   perl5.22.2.exe slic3r.pl --export-svg bracelet.stl --layer-height 0.02 --first-layer-height 0.02 --bottom-solid-layers 0 --top-solid-layers 0 

[Here are the stl and the output svg](https://github.com/alexrj/Slic3r/files/385449/Bracelet.zip)

lordofhyphens, 2016-07-27:
Thanks for checking.

lordofhyphens, 2016-07-27:
@alexrj Any thoughts as to what could be going on?

alexrj, 2016-08-03:
I can't test right now. But how close are those polygons?

ImInSpace, 2016-08-08:
More or less 0.1 millimeters