slic3r/Slic3r #3442: Non-planar FDM

lordofhyphens, 2016-07-27:
* http://hackaday.com/2016/07/27/3d-printering-non-planar-layer-fdm/
* https://github.com/makertum/non-planar-layer-fdm

Is this something we should look into bringing into upstream? 

Tagging @makertum to bring into conversation

Patola, 2016-07-28:
That... would... be... AMAZING!!!!! AWESOME! PLEASE DO IT (but first, custom supports to counter simplify3d)

Check out this project, it has non-planar slicing in java: https://github.com/nick-parker/Bread

makertum, 2016-07-28:
Great stuff, would love to see that feature!

I guess this can be achieved without massive changes in the slicing process itself. For static displacement maps, like the wavy cube (image below), I currently externally tesselate the STL geometry, pre-warp it, send them through the planar Slicer, then displace them in the post-processor.

Surface-geometry based displacement maps, like the wing (image below), could be generated from the STL file itself. It would be just a depth map of the top surface. The process of pre-warping, slicing and post-displacing stays the same.

All the best,
Moritz

**Cube:**
[image]

**Wing:**
[image]

FlameRunner, 2016-07-30:
That method obviously gives us ways to get some great prints. It would be nice to have it implemented in Slic3r, not even for some useful applications but just for the joy of experimenting itself.

On the other hand, I can't really imagine using it for regular everyday printing.

kefir-, 2016-07-30:
I imagine this would be a very useful feature. For angles within some limit, the typical 3d printed steps could be avoided, and printed parts could more accurately represent the original model. If it works well I don't see why it shouldn't be enabled by default with some conservative parameters?

burbilog, 2016-08-02:
Simply amazing. Please, do it!

rPman, 2016-08-07:
wow! this feature must be

mrvn, 2016-08-08:
The pictures show non-flat layers throughout the object. That's probably not even necessary. Instead print the inside normal with fixed thickness and stairs and all. But then the top fill layers you make non-flat, filling out the stairs in the process. Or a mixed mode. Allow non-flat layers throughout but when they get to thin allow stairs (on the inside).

mrjohnc, 2016-10-04:
Non-planar layers all the way through the print could offer other functions including offering increased layer bonding by making all layers 'wiggly' which would make the layers stronger in the x-y plane. 

Non-planar layers may also allow other possibilities when printing with multiple materials including coating one material with another.

alexrj, 2016-11-24:
Hey @makertum!
Sorry for jumping into the conversation so late. Your script is very nice, and it would be cool to include it in Slic3r so that more people can benefit and experiment. The hardest part is making a visual preview of the deformed output...

dmitryprokopchuk, 2017-05-27:
Incredible! It must be in the Slic3r !!!

probonopd, 2019-08-01:
For reference:

> CurviSlicer: Slightly curved slicing for 3-axis printers
> 
> Abstract : Most additive manufacturing processes fabricate objects by stacking planar layers of solidified material. As a result, produced parts exhibit a so-called staircase effect, which results from sampling slanted surfaces with parallel planes. Using thinner slices reduces this effect, but it always remains visible where layers almost align with the input surfaces. In this research we exploit the ability of some additive manufacturing processes to deposit material slightly out of plane to dramatically reduce these artifacts. We focus in particular on the widespread Fused Filament Fabrication (FFF) technology, since most printers in this category can deposit along slightly curved paths, under deposition slope and thickness constraints. Our algorithm curves the layers, making them either follow the natural slope of the input surface or on the contrary, make them intersect the surfaces at a steeper angle thereby improving the sampling quality. Rather than directly computing curved layers, our algorithm optimizes for a deformation of the model which is then sliced with a standard planar approach. We demonstrate that this approach enables us to encode all fabrication constraints , including the guarantee of generating collision-free toolpaths, in a convex optimization that can be solved using a QP solver. We produce a variety of models and compare print quality between curved deposition and planar slicing.

https://hal.archives-ouvertes.fr/hal-02120033

probonopd, 2019-08-22:
There is a Cura fork that implements nonplanar slicing:
https://github.com/Zip-o-mat/Slic3r/tree/nonplanar

psavva, 2019-08-23:
Just a pull request is needed :)

https://github.com/Zip-o-mat/Slic3r/tree/nonplanar-thesis

raylove518, 2019-09-04:
I have completed the installation according to the tutorial. Now the problem is that when Ubuntu opens the slic3r software, there is no option for nonplanar layers. Why is this? @makertum

Clem-, 2019-09-04:
> There is a Cura fork that implements nonplanar slicing:
> https://github.com/Zip-o-mat/Slic3r/tree/nonplanar

You said **Cura** fork?

probonopd, 2019-09-04:
Typo. Corrected.

AvinashPudale, 2019-09-21:
We need non planner fdm feature in modifier meshes.

TQwan, 2019-09-28:
> Is this something we should look into bringing into upstream?

Yes, please 👍

chiboreache, 2020-04-09:
:point_up:

n8bot, 2020-07-11:
Ahem https://github.com/mfx-inria/curvislicer

nesdnuma, 2020-07-14:
> Ahem https://github.com/mfx-inria/curvislicer

Quite interesting but could binaries be provided? Everyone is not a master in compilation...

foreachthing, 2020-07-14:
> Quite interesting but could binaries be provided? Everyone is not a master in compilation...

Right there: https://github.com/mfx-inria/curvislicer/releases/

nesdnuma, 2020-07-14:
Great! Thanks a lot @foreachthing!

nyxaria, 2020-08-19:
Is this happening?

Ben-MSTR, 2021-01-22:
Hi,
First of all, thank you for this non-planar version.
I do succeed in printing desired non-planar top layers but I wonder if there is a tip for bottom layers too ?
I would love to print non-planar bottom layers on support.
Thanks for answers and helps.

arne182, 2021-05-31:
Would it be possible to get an updated binary for this?
@JuDePom 
@sylefeb
@Phazon54