slic3r/Slic3r #3446: extreme overextrusion in top/bottom layers

ErikHorn, 2016-07-31:
### Version
1.2.9 and 1.2.9-347-g34f853

### Operating system type + version
Windows 10 64-bit

### Behavior
When printing the attached object, I encountered extreme overextrusion starting in layers 2 or 3. 

The slicing settings for this object are what I'd typically use for other objects. The most notable differences from default settings is that I do not use automatic extrusion width.

Here is a picture of a prior test run to show the problem. The print was aborted shortly into layer 3. The first two layers printed ok.

[image]

Here are the first 3 layers from the preview tab. I would expect that layers 2 and 3 should be similar to layer 1.

[image]
[image]
[image]

This file contains the STL file, gcode output from slic3r, and the slic3r config file.
[stl_gcode_config.zip](https://github.com/alexrj/Slic3r/files/392629/stl_gcode_config.zip)

makertum, 2016-08-17:
I can confirm this, same issue on OSX 10.11.6, Slic3r 1.2.9. The issue (mysteriously) happens on some STL files, while others are unaffected. Eric's STL file results in the same behavior on my setup, mostly independent of the used settings. The issue seems to be linked to very narrow infill regions (like the narrow region where the dovetail triangles are attached to the object) and can be mitigated by reducing the number of perimeters (effectively leaving more space for the infill).

alexrj, 2017-03-17:
This is fixed as of 1.3.0-dev. Thank you for reporting!

[image]

closed 2017-03-17 by alexrj