slic3r/Slic3r #3448: Poor extrusion width defaults

KevinOConnor, 2016-08-02:
On slic3r v1.2.9 (Fedora Linux), I am seeing surprising default extrusion widths.  This is on a printer with a 0.35mm nozzle and with all print settings set to default.  When producing gcode the following is placed at the top of the .gcode file:

; external perimeters extrusion width = 0.35mm
; perimeters extrusion width = 0.39mm
; infill extrusion width = 0.39mm
; solid infill extrusion width = 0.39mm
; top infill extrusion width = 0.39mm

At first glance, these settings look to be sane.  However, upon inspection of the gcode move commands (after the initial layer), the actual line spacing between internal extrusions is 0.321 and spacing between internal/external perimeter extrusion is 0.303.

It appears the discrepancy between reported extrusion widths (0.35/0.39) and extrusion spacing (0.303/0.321) is due to slic3r's estimate of extrusion "bowing" as described at: http://manual.slic3r.org/advanced/flow-math

However, this means that by default as slic3r extrudes adjacent paths the nozzle opening will be in overlap of the nozzle opening of previous paths.  That is, choosing a default extrusion spacing (0.303/0.321) smaller than the nozzle width (0.35) is surprising and questionable.  I expected slic3r to choose a default extrusion spacing width no smaller than the nozzle size.

At the other extreme, if I set the layer height in slic3r to 0.05mm and leave all other print settings at their default, then I see this at the top of the gcode file:

; external perimeters extrusion width = 0.35mm
; perimeters extrusion width = 0.59mm
; infill extrusion width = 1.93mm
; solid infill extrusion width = 0.59mm
; top infill extrusion width = 0.59mm

An infill extrusion width of 1.93mm is surprising and questionable on a 0.35mm nozzle.  I expected slic3r to choose default extrusion widths of no larger than about 180% of the nozzle size.

Hasi123, 2016-08-06:
I also have a similar issue when I set the extrusion width to auto. The extruder pushes out too much plastic and walls get so irregular, that the part gets unusable. 
[image]

lordofhyphens, 2016-08-15:
@Hasi123 turn off gap fill or use 1.3.0-dev.

lordofhyphens, 2017-03-11:
See #2693 for more pictures. 

https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/Flow.cpp#L91 defines the automatic width, which has no limit on extrusion width for internal infill, and as you can see, the algorithm tries to make the traces wider to fit a flow. That height^2 term (and nozzle_diameter^2) is the general cause of the absurd layer widths as far as I can tell.