slic3r/Slic3r #3449: Allow setting extrusion width spacing instead of "targeted extrusion width after estimated bowing"

KevinOConnor, 2016-08-02:
On slic3r v1.2.9 it is possible to manually override slic3r's choice of default extrusion widths.  This is done in the "Print Settings->Advanced->Extrusion width" section.  However, it appears these settings do not directly set the extrusion spacing nor does it directly set the extrusion volume.  Instead, it appears these settings specify a "target extrusion width after estimated extrusion bowing" - for example, if one sets an extrusion width of 0.6mm with a 0.3mm layer height then slic3r uses a formula (described at http://manual.slic3r.org/advanced/flow-math ) to obtain an extrusion spacing width of 0.536.  The volume of extrusion is then proportional to this calculated extrusion spacing width (eg, volume = line length * 0.30 * 0.536).

This is a request to change slic3r to allow the user to directly specify extrusion spacing width (and thus directly specify extrusion volume proportional to layer height) instead of setting a target extrusion width after estimated bowing.

There are several reasons why I think directly setting the extrusion spacing width is superior:

1 - the extrusion width after bowing is a "fuzzy" estimate that may be inaccurate on some printers.  In particular, the actual bowing may change depending on the type of plastic, the temperature of the nozzle, the speed of the printer, the layer height, etc.  In contrast, the extrusion spacing width and extrusion volume can be controlled with high precision on the printer.  It would be preferable to allow users to set a "hard parameter" instead of specifying a fuzzy "target parameter".

2 - setting the extrusion spacing is genuinely more useful.  For example, if a user has a part with many 2mm walls, the user may wish to set an extrusion spacing of 0.5mm to ensure exactly 4 tracks for those walls.  The current slic3r settings would likely mislead the casual user into setting non-ideal widths.

3 - the formula for determining how slic3r calculates extrusion spacing is complex and non-obvious. Thus, users (like myself) that wish to specify an extrusion spacing need to do extended research and involve a calculator to accomplish the goal.

4 - the current formula results in extrusion spacing width that changes with different layer heights.  This is particular painful for users that wish to test the quality differences between different layer heights - as each layer height ends up changing all the tracks.  Ideally changing the layer height would change just the layer height (and extrusion volume proportionally to layer height).

lordofhyphens, 2016-08-17:
This should be rather straightforward to aaccomplish. After all there are functions to calculate the flow amount from the spacing.

However, I am not seeing hard examples of why the current approach is incorrect or fails (or is inferior to the proposed method). Without those compelling arguments it will be difficult if not impossible to convince @alexrj to accept a pull request, much less work on it himself.

FlameRunner, 2016-08-25:
Maybe just some super-expert mode, allowing us to directly input line spacing and flow speed/head speed ratio?