slic3r/Slic3r #3450: Overhang detection and cooling activation

Tinchus2009, 2016-08-02:
### Version
latest 1.3.0 DEV

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win10 64

### Behavior
When printing an overhang we have some issues: the borders curl up. We fight this useing air cooling. In my case, I have a really good air layer cooling system, I can print overhangs at 20 degrees with no problem at all. But, only if I have my fan at full power. And I can do this when useing PLA, imposible to do if I use ABS. In the case of ABS, when I see an overhangs come, I manually use cooling and once the overhang area finishes I turn it off.

_Is this a new feature request?_

The feature request: a detection of everhang areas and so the ability to trigger the layer cooling there. In that way, we could control the cooling fan when we need it, and in case of ABS this can be critical.

VanessaE, 2016-10-26:
In addition to that, bridging speed and flow ratio desperately need to be invoked around overhangs also, especially if the overhang has a small perimeter.  There's only just so much one can do with layer fans.

bubnikv, 2016-10-26:
I believe if you enable the "Detect bridging perimeters", you will get the bridge flow and fan on at overhangs.

VanessaE, 2016-10-26:
Not in this particular case. Give this model a try: http://www.thingiverse.com/thing:1849506  Basically every "stick" this object is made from is at a steep angle, probably 25 or 30 degrees from the horizontal.

No matter what settings I fiddle with, I can't get Slic3r to slow down around those overhanging perimeters (bridging speed is set to 10 mm/sec, compared to my "small perimeters" speed of 30 and a regular speed of 60), or change the flow rate (should be 0.5), or really anything else that would help them print without supports or curling up.  I already had the fan set to 100% for the whole print, however if I set the minimum to 50% and export, I can see in the g-code that it is switching to 100% frequently, so overhangs are being detected.

VanessaE, 2016-10-26:
A quick check in blender says those overhangs are 35 degrees from the horizontal.

VanessaE, 2016-11-01:
On second thought, forget bridging settings.  Slic3r desperately needs separate settings for overhang speed and flow ratio so that one can configure it to go super slow and under-extrude without it affecting normal bridging.