slic3r/Slic3r #3452: [request] auto arrange part distance

Lord2Vader, 2016-08-05:
### Version
_1.2.9_

### Operating system type + version
_Windows 10_

### Behavior
* _when using arrange stl parts or objects are mostly far away from each other. Parts are further away than they need to be. This is problematic with flexible filaments and filaments where retraction does only show a great effect over an specific distance.

Also L shaped parts are arranged blockwise. Interleaving them would be a nice feature (maybe an option to auto rotate_

_Feature request_

lordofhyphens, 2016-08-12:
Near as I can tell, the value is given by min_object_distance() in PrintConfig.cpp
This is the maximum of a parameter called duplicate_distance that doesn't seem to have a GUI parameter to tweak it or extruder_clearance_radius (if you have complete_objects on, it's the max of the two). 

Add the following line to Tab.pm at line 1047 to see if this is useful to you.
            $optgroup->append_single_option_line('duplicate_distance');

xs/src/libslic3r/Model.cpp is where the rest of the arrange logic lives, feel free to play around with it.

Rotate is covered by other issues.

foreachthing, 2017-05-30:
@lordofhyphens, where would I add this option (to the GUI) to a Windows version of Slic3r (1.3. dev), if **I can't compile**?

lordofhyphens, 2017-05-30:
It would be in the preset editor perl files,  under lib/Slic3r (relative to slic3r.pl)  in an automatic build.

Look around for a list of options in a qw () call and add the option name there. No recompilation needed.

The rest of the stuff I had been talking about I'd in libslic3r and would require a rebuild. If you do rebuild and need the change on a system that is not set up to build, cd to package/win in a Powershell prompt and run `compile_wrapper.ps1

foreachthing, 2017-05-30:
Thanks a lot!! So, I added this: `duplicate_distance` to _PresetEditor.pm_ at around line 453 (at the end of `sub options { return qw( `.
Then `$optgroup->append_single_option_line('duplicate_distance');` at around line 673 (after `$optgroup->append_single_option_line('resolution');`).
Now I've got this option in Print Settings -> Advanced -> Others.

lordofhyphens, 2017-05-30:
Have fun experimenting!

alexrj, 2017-05-30:
I think duplicate_distance should be removed from PrintConfig, not included in presets and moved to the application Preferences instead.

lordofhyphens, 2017-05-30:
Wasn't it used in cli?

foreachthing, 2017-06-01:
>  and moved to the application Preferences instead

I disagree. I don't think that this is an application setting. It's a "Print Setting", since it influences the output on the printer's plate. I'd never look in application preferences for such a parameter.

lordofhyphens, 2017-06-02:
@foreachthing It's a UI option, first and foremost, and it should be in the domain of the UI and not part of libslic3r unless we can come up with a really good reason why it should live there.

To your point of "it influences the output on the printer's plate"... Everything does, but that doesn't mean it should be part of libslic3r (the slicing library) instead of slic3r-gui (the frontend to libslic3r that the user sees).

zpeleg, 2018-07-01:
How can I add this option on the latest version? I can't find any of the files or lines mentioned.

exetico, 2019-11-09:
So, how did it go. Is it actually added in the GUI :D?

martincollar, 2019-11-11:
This would be really helpful!

foreachthing, 2019-11-11:
I didn't want to make a PR, since I disagree with the location of the setting. I don't think it's an application setting and @alranel  thinks it is and he's the boss ;-)
But that's an easy one, if everybody agrees on location ...

martincollar, 2019-11-11:
I have no voting rights :D but still i would also say this is Print settings not application..

supermerill, 2019-11-19:
To be sure to complicate thing, i vote for a printer setting because it depends more on the extruder/nozzle diameter/build plate.

;-p

the78mole, 2020-01-02:
Unfortunately, if I add above suggestion to 
Slic3r-1.3.0.64bit\lib\Slic3r\GUI\PresetEditor.pm
at line 703

        {
            my $optgroup = $page->new_optgroup('Other');
            $optgroup->append_single_option_line('xy_size_compensation');
            $optgroup->append_single_option_line('resolution');
            $optgroup->append_single_option_line('duplicate_distance');   # This line added
        }

Slicer crashes, when I try to enter the print settings dialog. I'm running Windows 10.