slic3r/Slic3r #3455: Selecting "About Slic3r" menu option exits application

fbrier, 2016-08-07:
### Version
1.2.9-3.fc23 x86_64 (Fedora)

### Operating system type + version
Fedora Linux 23 (up to date)

### Behavior
* _Describe the problem_
Launch Slic3r from the Graphics menu.  The app launches quickly.  All tabs work and even dialogs pop up properly, such as "Load Config...", or "Configuration Assistant...".  Selecting, from the Help menu, the "About Slic3r" menu option causes the application to instantly close its main window, presumably due to crashing.

* _Steps needed to reproduce the problem_
Launch the application
Select "About Slic3r" menu option

  * _If this is a command-line slicing issue, include the options used_
Not applicable

* _Expected Results_
An about dialog box showing the version number?

* _Actual Results_
Application window closes
This occurs every time (very reproducible)

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
Not applicable

Notes:
This is a fairly powerful machine with an 8 core AMD processor and 32GB of memory.  This was  a clean install with no mods loaded.  I performed a reinstall, but the problem still exists.

lordofhyphens, 2016-08-07:
Try with 1.3.0-dev

fbrier, 2016-08-07:
I made the assumption that to try 1.3.0-dev, I needed to clone the repo and build master.  After following the directions for Fedora, and installing in a few missing packages, like boost and boost-devel, I finally got it to build, with lots of warnings, mostly about [-Wreorder], but it is failing ALL the tests and subtests, and then skipping a bunch at the end, so obviously all is not well.  It starts with this error:
```
t/angles.t ..................... Subroutine Slic3r::Config::GCode::new redefined at /home/fbrier/Projects/Slic3r/t/../lib/Slic3r/Config.pm line 390.
Bareword "Slic3r::Config::Static::new_GCodeConfig" not allowed while "strict subs" in use at /home/fbrier/Projects/Slic3r/t/../lib/Slic3r/Config.pm line 390.
Compilation failed in require at /home/fbrier/Projects/Slic3r/t/../lib/Slic3r.pm line 38.
BEGIN failed--compilation aborted at /home/fbrier/Projects/Slic3r/t/../lib/Slic3r.pm line 38.
Compilation failed in require at t/angles.t line 12.
BEGIN failed--compilation aborted at t/angles.t line 12.
t/angles.t ..................... Dubious, test returned 255 (wstat 65280, 0xff00)
No subtests run 
```
And ends with this
```
t/vibrationlimit.t           (Wstat: 65280 Tests: 0 Failed: 0)
  Non-zero exit status: 255
  Parse errors: Bad plan.  You planned 9 tests but ran 0.
Files=33, Tests=0,  2 wallclock secs ( 0.12 usr  0.07 sys +  2.11 cusr  0.41 csys =  2.71 CPU)
Result: FAIL
Some tests failed. Please report the failure to the author!
```

lordofhyphens, 2016-08-14:
I can't reproduce this on Debian or the Linux build from Slic3r.org. If you're working with whatever Fedora is packaging into their distro, you should contact that package maintainer.

lordofhyphens, 2016-08-14:
Tests aren't failing on either of our CI servers, nor my own system. I suspect the actual fault lies in your system setup somewhere. Make sure you aren't using Perl 5.16.

Closing :(

closed 2016-08-14 by lordofhyphens

fbrier, 2016-08-15:
Perl v5.22.2.  Thank you for your assistance.

lordofhyphens, 2016-08-15:
You should also check the official 1.2.9 build from slic3r.org. if it is crashing too, there is something going on with your libraries. If it does not crash, the package you were using probably has something going on with it (and should be reported to the package maintainer).