slic3r/Slic3r #3456: Feature Request: Adding multicolor Extruder with mixing cabability

arget888, 2016-08-11:
Hey,
would it be possible to consider the ability for multiextrudign with a single hotend end and Color mixing like ist possible with the Diamond hot end from reprap.me, i think that would be a gread Feature?

lordofhyphens, 2016-08-11:
Multiple extruders/colors are already doable, not sure what you are looking
for here

arget888, 2016-08-11:
I am asking for the posibility to mix colors from 2 or 3 extruders in a singel hot end like the diamond hot end does.

lordofhyphens, 2016-08-12:
I'd *probably* look at a pull request regarding it, but since I don't have (and don't need/want/use) one... 

It'd probably require refactoring of the extruder configuration to start and a gcode flavor with special options to do replace a single extrude move with 3 moves instead in the correct amounts, and some way to define how to get to color X from A+B+C, probably involving some color theory and RGB assignments for the three filaments. That's the easy part.

Then you'd have to work out how to get those colors assigned to the regions of the print. Currently that's done by assigning filament profiles to regions. I believe this information can be saved by Slic3r to AMF.

Patola, 2016-08-12:
RGB? Wouldn't it be addictive, like CMYK?

bubnikv, 2016-08-12:
As far as I am known, only Repetier firmware supports mixing a extruder. Or
is there any support in Marlin as of today?

I believe the way it has been implemented in Repetier firmware is through
virtual extruders and by setting a color palette for each of the virtual
extruder in the firmware.

I suppose to support a mixing extruder (currently I am aware of the Diamond
as the only specie available), effort in the slicer shall be coordinated
with firmware development.

Vojtech

lordofhyphens, 2016-08-12:
I think what was being asked was to generate gcode so that the virtual
extruder setup was unnecessary.

bubnikv, 2016-08-12:
But for which firmware? Who will interpret this G-code?

arget888, 2016-08-12:
I am using marlin firmware, shouldn't it be possible to use 2 or 3 extrudera at the same time, like to 3 axis at the same time?

bubnikv, 2016-08-12:
Marlin surely does not support it, at least the Marlin from the start of
this year, which we forked at Prusa3D.

mrvn, 2016-08-25:
There are actually 2 things to support:

1. A hotend with single heater but 3 extruders. On tool change the old filament has to be purged. This could be done on a wipe tower outside the object or the tool change could be done ahead of time while printing infill. The former is needed when printing different materials, e.g. water soluable support, the later can be done when it's just a color change.

2, Mixing colors. I think it makes most sense there to have virtual extruders with an E1/E2/E3 ratio setting that gives the desired color. Again old filament needs to be purged to achive color correctness on switch. From what I heart the colors don't mix well though and it's more like multi colored toothpaste with regions of individual colors in the extrusion. Virtual extruders could be converted to colors in post processing, slic3r just needs to support many extruders here.

arget888, 2016-09-14:
The Marlin 1.1.0 RC7 has the feature of a mixing extruder. If I a understand it correctly, the slicing software has to support generating gcode for mixing filament or am I wrong?

lordofhyphens, 2016-09-14:
The way that Repetier supports it iirc is through virtual extruders, so all
the slicer has to do is say "use extruder 38484" and the firmware handles
the rest.

Handling it in the slicer means working out how much to move all N axes to
get what is desired.

That is probably not going to happen, at least from myself or @alexrj.
Especially since the names of the axes is going to change, etc.

I would probably look at a pull request but it is going to touch quite a
bit of the interface.

arget888, 2016-09-14:
Couldn't it be done by mixing the colors in an easier way like we do it at entering colors on the pc. For e.g. if 1meter of a orange color is needed the programm can calculate that for e.g 50%red and 50% of yellow filament is needed and than it says extrude 0.5 meter of red and 0.5 meter of yellow?

lordofhyphens, 2016-09-14:
That is going to be firmware dependent for the gcode. As far as I know,
neither @alexrj or I own one of these hotends either.

So talking about it is all well and good but that isn't going to get me (at
least) to spend time working on it. Hence why I said I would review a pull
request if someone else wants to take a crack at it.

lordofhyphens, 2016-09-14:
From a UI standpoint we'd need a good way of defining and using N extruders (better than what we have now). At minimum the color information would need to be in the file format as well (so we're at AMF required).

arget888, 2016-09-14:
ok thx, i hope some else also think it would be nice and can do it ;)