slic3r/Slic3r #3457: Pressure advance "unbalanced"

VanessaE, 2016-08-14:
Sorry if the title seems obscure, I couldn't think of a better way to describe this issue.  I am currently at 8a72ffb302bf4cb733fe5a5eaaba8e5222681179 in Slic3r's repo, and have been using pull request #3287 (29fbe89b0b018b19f2183c8fd981df4cbbfd52b3 et al.).

I'm finding that pressure changes are being "overcompensated" for in some situations.

Use your favorite test object to tune your pressure advance to *just* eliminate the blobs that usually occur on high-to-low pressure changes associated with switching between internal and external perimeters.  With a little luck, you should still be able to see where the blobs would have been.  Then give this Thing a go: http://www.thingiverse.com/thing:289516 (you can pick any size).  This part is, in essence, shaped like a drinking cup (a *very* leaky one :stuck_out_tongue:)

Of particular note though: the bottom-inside of the "cup" is completely flat, so Slic3r uses the "top solid infill" scheme when it does the upper-most layer of it.  That part is thin enough that, at least on my setup, only solid infill is used throughout.  All other layers beyond that surface are just walls consisting entirely of internal/external perimeters and some gap fill.  

When printing, observe the first 10 or 15 mm of the first wall section that gets printed right after the first layer change after the bottom-inside is done.  On my setup at least, the wall there is being starved - probably the first internal perimeter, to be specific.  The effect "bleeds through" to its neighbors due to the rough surface there.

The more you tune out the internal-to-external blobs, the worse this "starved" effect gets.  It is as though the program's pressure advance algorithm is expecting top solid infill to be higher pressure than it actually is.

Config bundle: [Slic3r_config_bundle_1.3.0.ini.zip](https://github.com/alexrj/Slic3r/files/416983/Slic3r_config_bundle_1.3.0.ini.zip)

lordofhyphens, 2016-08-17:
Have you used autospeed instead of pressure advance? Just a thought; autospeed is supposed to maintain constant pressure throughout the print.

VanessaE, 2016-08-17:
I have used it before and found it not to be too workable for me (mostly a couple of things print too fast that ought, in my opinion, to go more slowly).  I do everything with "normal" speed settings (typically maxing out at 60 mm/sec), and not in volumetric mode.

bubnikv, 2016-08-17:
I am working on a generalization of the constant volumetric pressure feature: A post-processor controls the rate of change in volumetric pressure, not the volumetric pressure itself. The idea is to let the volumetric pressure grow and fall slowly than usual. So far it looks promising, but sometimes Slic3r generates extrusions with weird extrusion rates, which break the logic. The problem of all the pressure control routines is, it breaks once the firmware starts limiting the moves. So the correct pressure correction algorithm has to know and simulate the particular firmware, or to be implemented into the firmware.

lordofhyphens, 2016-08-17:
Both approaches have their pitfalls of course :(. The slicer doesn't
necessarily know how the firmware will do things and the firmware has to
take a relatively short view of the problem.

I suspect that a good, novel, volume algorithm is publishable in a
conference or journal, btw. And if it has been published that's unrefutable
prior art :)

lordofhyphens, 2016-08-27:
Ran across https://drive.google.com/file/d/0B5UvosQgK3adaHVtdUI5OFR3VUU/view while looking at something else, apparently it's implemented in Marlin now.

VanessaE, 2016-10-14:
After a ton of hair-pulling and a LOT of wasted filament, :frowning_face: I've decided that Marlin's Linear Advance feature just does not work well for me - it's too aggressive for my poor extruder. :stuck_out_tongue:

Meanwhile, I think the slicer is the better place for this for one reason above anything else:  There's no way for the firmware to know anything about the object itself, so it can't plan any further ahead than the contents of its input buffer.  All it sees are a dozen or so incoming line segments at a time, but it can't know, for example, that a particular group of lines represents infill versus a perimeter, so it has to assume every incoming line segment has to be compensated for.  The result is that the extruder goes nuts trying to keep up.

Slic3r's feature would be just fine I'm sure, if it reacted to the speeds created by its "slow down" feature.  I'm not saying to change the algorithm, just that "slow down" should be taken into account just the same as if the user had entered equivalent speeds in the "Speed" section of the "Print Settings" tab.

VanessaE, 2016-12-30:
Some new results, after a lot more testing.  All prints below were done with identical settings, under identical conditions, with pressure advance enabled (`K=0.05`), and with only the "slow down" time varied, as noted below.  I'm at commit 43c62d468c79b6ba8b956bcd0f4ccbeb72f6454d now, by the way.

Slow-down threshold set to 5 seconds (the default,  low enough to effectively disable it)
[image]
Corresponding g-code: [ctrlV_3D_test-v3--5s.gcode.zip](https://github.com/alexrj/Slic3r/files/678228/ctrlV_3D_test-v3--5s.gcode.zip)

Threshold set to 15 seconds.
[image]
Corresponding g-code: [ctrlV_3D_test-v3--15s.gcode.zip](https://github.com/alexrj/Slic3r/files/678230/ctrlV_3D_test-v3--15s.gcode.zip)

Threshold set to 45 seconds.
[image]
Corresponding g-code: [ctrlV_3D_test-v3--45s.gcode.zip](https://github.com/alexrj/Slic3r/files/678229/ctrlV_3D_test-v3--45s.gcode.zip)

As you can see, as the slow-down feature bumps up the layer time, pressure advance goes kinda wonky (look for example at how the shapes of those overhangs left of the dome change in the "45 second" print).  Other artifacts can be seen on the spire to the left of those, and the pyramid left of it, as well as on the thin walls on the other side of the object (out of focus in these photos).

Of particular note, though, is the ridge across the back of the large dome in the "45 second" print.  This is not a one-time artifact; it will appear consistently as long as the slow-down threshold is set high enough.  Anyway, my calipers put this ridge roughly at 16.6mm.  Meanwhile, gcode.ws shows that the slow-down feature induced a layer time increase in the same vertical area, from 10-13 seconds up through layer 80, to 25.6 seconds on layer 81 and 35.8 seconds on layer 82.  That suggests to me that the slow-down feature caused the ridge.

The dome is featureless (aside from low-poly faceting), and becomes the only thing to print after the spire is done.  Since that ridge occurs well above the top of that spire (about 1.2mm higher), there's no obvious, visible reason for it to happen.  There are similar artifacts in many other places in this print, but no such problems with the other two prints.

In the "5 second" version, gcode.ws shows a more-or-less smoothly-varying layer time for the dome above the spire, starting at 6.4 seconds just after the spire.  There's only a few tenths of a second difference here or there between layers 80-84, and indeed the preview pane in Slic3r shows nothing unusual at all about the sliced result in this region.

Even with lots of fans/airflow, and only a moderate printing temperature of 205°C, the plastic still needs time to cool, and with this model, layers are simply laid down too frequently for the last several millimeters of the dome if slow-down isn't invoked, causing it to warp from the excess heat (note that this is not what causes those "serrated" edges, that's something else entirely).

The model under test is http://www.thingiverse.com/thing:1363023.
Config bundle: [Slic3r_config_bundle_1.3.0-dev-2016-12-30.ini.zip](https://github.com/alexrj/Slic3r/files/678319/Slic3r_config_bundle_1.3.0-dev-2016-12-30.ini.zip)
(use "Single material, 60 mms, 0.2mm layers", "Atomic PLA, Royal Blue", and there's only one Printer profile)

VanessaE, 2017-04-18:
No action from anyone besides myself since August last year, and this specific problem doesn't manifest anymore.

closed 2017-04-18 by VanessaE