slic3r/Slic3r #3458: Weird seam positioning

VanessaE, 2016-08-14:
### Version
 I am currently at 8a72ffb302bf4cb733fe5a5eaaba8e5222681179 in Slic3r's repo, and have been using pull request #3287 (29fbe89b0b018b19f2183c8fd981df4cbbfd52b3 et al.).

### Operating system type + version
Debian

### Behavior
Slic3r is choosing completely weird places to put the seams in some situations.  By weird I don't mean that it just chooses the wrong stack of vertexes - rather, it distributes the seams somewhat, but not completely randomly, even though I'm using "Aligned" mode.  "Nearest" mode also exhibits this.  The following images should explain it better.

The object as it renders in Pronterface (because you can see the perimeters' start/endpoints easier):
[image]

And as it looks when printed:
[image]

As you can see, where Pronterface shows a difference in shading, the printed part has a layer transition there.  Note that when this object was printed, my pressure advance setting was sub-optimal, so the blobbing here is kinda serious, but neither that nor any other settings seem to affect this weird "dispersed" pattern, other than if I put the seam position to "Random" (at which point, it does randomize them).

#### STL/Config (.ZIP) where problem occurs
Model file: http://www.thingiverse.com/thing:289516 (the 40mm size, but it doesn't matter)
Config bundle: [Slic3r_config_bundle_1.3.0.ini.zip](https://github.com/alexrj/Slic3r/files/416983/Slic3r_config_bundle_1.3.0.ini.zip)

VanessaE, 2017-03-14:
The recently-added "back" seam alignment option is enough to avoid this issue, at least for the same model.

alexrj, 2017-05-01:
Thank you for this great catch. I fixed it.

closed 2017-05-01 by alexrj