slic3r/Slic3r #3459: Over-extrusion when odd number of vertical shells required

centripetal3, 2016-08-15:
### Version
Slic3r 1.2.9

### Operating system type + version
Ubuntu Linux 14.04 LTS

### Behavior
* When producing a vertical wall with an odd number of shells, a single line of infill is produced which is larger than the volume it's intended to fill. The extra extruded plastic builds up and eventually collides with the nozzle, causing the print to fail. When a vertical wall has an even number of shells, the extrusion does not use any infill and the part prints flawlessly.

I'm not sure if I'm doing something wrong, or if this is possibly related to https://github.com/alexrj/Slic3r/issues/3448

#### STL/Config (.ZIP) where problem occurs
[testcase.zip](https://github.com/alexrj/Slic3r/files/417450/testcase.zip)

[image]
[image]
[image]
[image]
[image]
[image]

lordofhyphens, 2016-08-15:
That single line is gap fill, and it's known to have overextrusion problems in 1.2.9. 
Suggest either using 1.3.0-dev from source or disabling gap fill (set speed of gap fill to 0).

closed 2016-08-15 by lordofhyphens

centripetal3, 2016-08-16:
Worked like a charm. Thank you!