slic3r/Slic3r #3460: Command line issue with rotation

xyReaper, 2016-08-16:
### Version
_1.2.9_

### Operating system type + version
Windows 7 64bit

### Behavior
This is an issue regarding the command line.
When the argument "--rotate" is used Slic3r seems to be interpreting it in radians rather than degrees.
below is an image with --rotate 0
[image]

here is the same file this time with --rotate 3
[image]

3 (in rad) is about 171 degrees which is what appears to be the rotation applied.

The problem is that it doesn't let you input the angle with a decimal so using it with radians is useless.
Can this be fixed please? And if possible, allow for decimals (even in degrees).

lordofhyphens, 2016-08-16:
I suspect it's related to the UI problem (no decimal), but I'll take a look.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-08-16:
Not sure what's going on here, as the rotate() command looks like (at least in 1.3.0-dev) it's coming from https://github.com/alexrj/Slic3r/blob/master/xs/src/libslic3r/Model.cpp.

Could you try this build of [1.3.0-dev](https://dl.bintray.com/lordofhyphens/Slic3r/:slic3r-fix-combine-infill--7faa0fb.zip) (you should be able to use the batch file the same as the exe if I set it up right).

xyReaper, 2016-08-16:
Same problem...
Am I doing something wrong?
I am simply running this

"slic3r.dev filename.stl -rotate 3 output.gcode"

lordofhyphens, 2016-08-16:
You're probably not doing anything wrong so far as I can tell from the
usage, etc. Also, I'm not seeing anything that would block you from passing
in decimals.

xyReaper, 2016-08-17:
It gives me the --help as in it didnt recognize the command.
Well, the problem can be fixed either by changing the radians to degrees or by allowing the decimal.
in the help it says that:
   --rotate            Rotation angle in degrees (0-360, default: 0)
Maybe i was implemented to take in integers only?

lordofhyphens, 2016-08-17:
https://github.com/alexrj/Slic3r/blob/master/slic3r.pl#L47 replace "=i" with "=f" and see how that goes.

xyReaper, 2016-08-17:
Nice! 

Thank you very much

lordofhyphens, 2016-08-17:
Reopening until I can fix the other related UI bug.

lordofhyphens, 2016-08-19:
The main problem is that while the underlying code is perfectly fine to interpret decimals, the UI widget for the GUI used apparently can't deal with floating point.

alexrj, 2016-12-16:
Thank you @xyReaper, I fixed that.
The UI thing is the same as #1617, so closing this one...

closed 2016-12-16 by alexrj