slic3r/Slic3r #3461: Object placement on Delta

frankvdh, 2016-08-17:
### Version
1.2.9

### Operating system type + version
Windows 7

### Behavior
To place an object, it appears that Slic3r v1.2.9 is minimizing the distance from the centre in the X and Y axes? That isn't the best strategy for a Delta... worst case, it leaves corners of the object extending outside the bed. This is really irritating because Slic3r doesn't allow me to manually position objects.

[capture](https://cloud.githubusercontent.com/assets/6589024/17721480/96207ef0-647e-11e6-88b8-d3572179792b.PNG)

In my view, for a Delta, the object should be placed such that  the smallest circle that contains all of the object is located at 0,0.

As a workaround, rotating the object around the Z axis changes its X & Y lengths, so it changes the placement in Slic3r... by rotating this object by 60 degrees, I can make Slic3r fit it on the bed.

Possibly this is related to issue #1544 ?

#### STL/Config (.ZIP) where problem occurs
[Config+STL.zip](https://github.com/alexrj/Slic3r/files/421584/Config.STL.zip)

lordofhyphens, 2016-08-17:
File -> Preferences menu, turn off auto-center. This gives you manual control over placement.

lordofhyphens, 2016-08-17:
Summary of feature request by @frankvdh 
> * Alternate arrangement method is desired for single component placement such that the part is contained within a circle situated at 0,0 with minimum radii. 
>     * Arrangement method is allowed to rotate parts about their center in Z. 

Somewhat related in scope to #2991

alexrj, 2016-11-24:
Interesting point.

We would need to center the centroid of the convex hull instead of the center of the bounding box. Easy to do, but we need to cache the centroids at load time.