slic3r/Slic3r #3462: GUI and command-line produce different results

el-bart, 2016-08-17:
### Version
1.2.9

### Operating system type + version
* Linux - Debian/testing
* amd64

### Behavior
when slicing file from command line ("slic3r --load config.ini center.stl") custom bed size/center are not applied.

when slicing from GUI ("slic3r --load config.ini --gui center.stl" -> Plater -> Export G-code) output is generated as expected.

this renders command line useless for automation, as it always generates wrong output. the only way to go is to slice via GUI mode.

#### STL/Config (.ZIP) where problem occurs
[example_issue.zip](https://github.com/alexrj/Slic3r/files/423430/example_issue.zip)

lordofhyphens, 2016-08-18:
If you want to center the print about some particular point on the bed in absolute coordinates, use --print-center=x,y

So if your print center from the config puts it in the center of the bed, use --print-center=0,0 for now.  I don't think the CLI even pays attention to bed_shape (and I don't think it has a reason to; the bed shape is just there as a UI guide for the user).

lordofhyphens, 2016-08-18:
And unfortunately print_center isn't a "real" PrintConfig option, so you can't just toss it into the config file.

el-bart, 2016-08-25:
sorry for lagging - been AFK for a few days.

this config is one i've generated about a year ago AFAIR. maybe it was an option back then? dunno... any way it's a very useful thing in real life. i've noticed this miss-alignment, when print size was almost of a bed size, and got a broken print.

i think it would be great to keep UI and CLI in sync, when it comes to config. i.e. the same config file should give the same output, regardless of UI/CLI generation mode.

lordofhyphens, 2016-08-25:
They are so far as I know, except for a few options that are CLI specific.

lordofhyphens, 2016-08-25:
This particular issue stems from the fact that the bed size doesn't actually factor into the slicing. I believe the plater basically supplies print-center to the engine.

el-bart, 2016-08-25:
perhaps having a single configuration representation inside the system, regardless of its origin, would do the trick? this would also naturally keep CLI's and GUI's capabilities consistent in future, when new features will be added. IMHO this would be a nice engine-UI separation, via an interface.

this would also solve the issue here, as the print-center option would become a part of the configuration.

alexrj, 2016-11-24:
@el-bart, the GUI allows free positioning on the plate. That's not possible from CLI...
Maybe the newly added `--dont-arrange` option fixes your needs?

Regardless of the GUI, can you describe your needs/problems in terms of automation?

el-bart, 2016-12-04:
@alexrj - thanks for the hint! it looks like this option will be just fine for my needs here. need not to print in the center, after all. ;)

for me GUI is almost a non-existing feature. :) i've used it to generate initial config file. nowadays i just edit it with text editor, when need some per-project changes and diff/merge between repos. slic3r is a part of my flow: OpenSCAD->slic3r->gcode. my Makefile template is here: https://github.com/el-bart/mini/blob/master/prusa_i3_xl/Makefile - so i basically just use OpenSCAD when designing and call make to generate a final output and/or preview a build result. the nice thing, is that i always get a repeatable output this way (developer's habits ;)).

alexrj, 2016-12-05:
As a recap, I confirm all slicing options are the same between CLI and GUI.
In addition, CLI has some options for replacing those commands that the GUI offers within its interactive interface, like model placement, scaling, rotation: while in the plater you can freely position your model in the bed, there's no friendly way for doing the same in CLI.

CLI offers two alternatives:
* if you can assume your STL files use the same coordinate system as machine/G-code, use `--dont-arrange`;
* if you don't trust your STL files coordinates, use `--print-center X,Y` and Slic3r will automatically center the print around that point.

closed 2016-12-05 by alexrj

el-bart, 2016-12-05:
i think it would be good to at least print a warning for options in a config file, that are not taken into account when used from CLI. status quo is a bit misleading. :/