slic3r/Slic3r #3464: Translation file PrintConfig.cpp, encoding

AndylgTom, 2016-08-18:
### Version
compile the repository 1.30....

### Operating system type + version
Windows10 x64_

### Behavior
Hi, help please.
Translate program to russian language. In the translation file PrintConfig.cpp faced with the problem of incorrect display of the translated text in the program. See screenshot [image]

That's a line 206 in file PrintConfig.cpp. Ie all the lines with "def->category =" appear in the gibberish. At that translated line "def->label =", def->tooltip = " displays the program normally.
How to fix this?
Thx.

lordofhyphens, 2016-08-19:
Check the encoding in the source files, otherwise your guess is as good as mine.

closed 2016-08-19 by lordofhyphens

AndylgTom, 2016-08-19:
How to understand you have to offer to convert Print Config.cpp UTF-8 to Unicode?
I tried - gives an error. And the problem with THIS file! (On the other files, such as Object.pm, Print.pm, converting from UTF-8 format to Unicode format helped. But that file PrintConfig.cpp it doesn't work - compile error.)
[image]
[image]

lordofhyphens, 2016-08-19:
Error message tells you, in C++ terms, what it thinks is wrong.

I am not going to offer to convert anything for you.

lordofhyphens, 2016-08-19:
Whatever program you used to process the text put a bunch of spaces and
null characters into the source code.

Clean it up and you will probably get somewhere.

AndylgTom, 2016-08-19:
So why in some russian text files in utf-8 format is displayed in the program interface is normal, and in the other case (Object.pm, Print.pm, PrintConfig.cpp) displaying gibberish? And in PrintConfig.cpp this applies only lines "def-> category". ???

Try to enter сyrillic text into the line 206 in file PrintConfig.cpp... And compile. I am sure at 100% it will be the same.

bubnikv, 2016-08-19:
Maybe not all the Slic3r code or the underlying Perl::Wx or the WxWidgets
library supports Unicode correctly?

AndylgTom, 2016-08-19:
> Maybe not all the Slic3r code or the underlying Perl::Wx or the WxWidgets
> library supports Unicode correctly?
I don't know

Apparently in this place source encoding UTF-8 displayed as Windows-1252. Why and how to fix this? 
[image]

AndylgTom, 2016-08-20:
Can delete the topic, the problem was Config.xsp и XS.c . Someone were missed encoding for categories.

lordofhyphens, 2016-08-20:
This isn't a forum, it is a bug tracker. Best I can do is lock the thread.