slic3r/Slic3r #3465: Infill and retraction

mjmeans, 2016-08-18:
### Version
1.2.9

### Operating system type + version
Windows 10 1607

### Behavior
The hexagonal infill causes a delay on every second 'row'. What I'm seeing is that instead of continuously going zig-zag row after row continuously, it will delay a brief amount of time before the next pair of hexagonal rows. And on some layers it will jump ahead two rows, then bag up one row, then ahead two rows, then back one row. This delay is probably caused by retraction settings for printers with Bowden tubes that require large retractions. On a 3 hour print of a large basically flat object at 20% infill, this substantially increases time to print and on 50% infill it is even worse. What I expect to see is a continuous zig-zag of hexagonal infill rows across the layer without jumping ahead and back and without retraction.

lordofhyphens, 2016-08-19:
50% infill with hexagon/honeycomb infill is generally overkill to start with. 

Also, please provide a copy of slic3r settings and a test STL as requested in the prompt and contributing guidelines.

Lord2Vader, 2016-08-20:
Have noticed that too.

Its a loop of two rows where it starts at the second row and prints on through the first, then jumps to the fourth and prints on through the third and so on.

landodragon141, 2016-10-21:
I can also confirm that this happens. It's every 3 layers.

mjmeans, 2016-10-22:
@lordofhyphens When your printing structural parts in ABS that will be exposed to both heat and high stress, (such as a high speed bearing carrier that will be mounted to PLA or other lower heat resistant parts), a 50% infill is a good balance between part strength and air insulation pockets to protect against heat transmission through the ABS part to more heat sensitive parts.

lordofhyphens, 2016-10-22:
@mjmeans 
>Also, please provide a copy of slic3r settings and a test STL as requested in the prompt and contributing guidelines.