slic3r/Slic3r #3471: Overhang threshold

claell, 2016-08-19:
### Version
1.2.9

### Operating system type + version
_Ubuntu 16.04_

### Behavior
* In expert settings there is a overhang threshold field for support. Default value is 0°.

It is interesting that this creates the same support as 45° does. Is 0° triggering something like a default value?
It would be more intuitive to put in 45° in the field then by default.
Also you might want to change the behaviour of support when putting threshold to 0° (I guess this has the same effect as disabling support generation).

lordofhyphens, 2016-08-19:
As the tooltip says, 0 is auto detection.

claell, 2016-08-20:
Thanks, did not read the tooltip and did not see the support changes for 0° when changing layer height for example.

closed 2016-08-20 by claell

bubnikv, 2017-01-27:
Auto-detect means "more than half of the perimeter width is not supported". This is not the same as a 45 degree overhang.