slic3r/Slic3r #3472: Slic3r crashes in PrintConfig.cpp due to a bug in the use of virtual inheritance, if compiled with Visual Studio 2013

bubnikv, 2016-08-21:
Please see the following commit:
https://github.com/prusa3d/Slic3r/commit/bd23fe9c44d4231066f1ceb5dea1b98143a1d95c

Fixed a crash in a constructor of FullPrintConfig due to an incorrect use
of virtual inheritance. Note that an invocation of ConfigBase::optptr()
is routed to FullPrintConfig::optptr() for all classes of the FullPrintConfig
hierarchy. FullPrintConfig::optptr() in turn invokes optptr()
of PrintObjectConfig, PrintRegionConfig, PrintConfig and HostConfig.
Due to the use of virtual inheritance, this all happens, when
PrintObjectConfig gets constructed as part of FullPrintConfig, but
at that time PrintRegionConfig, PrintConfig and HostConfig are not
constructed yet. Accessing them at that time leads to crashes,
when compiled with Visual Studio 2013 compiler. For some reason
the code generated by gcc does not crash, but I believe the behavior
is undefined and it is better to be fixed anyway.

The patch solves the problem by calling set_defaults() by the topmost
object, which not only fixes the crashes, but also avoids repeated
initialization.

On a similar note, due to the use of virtual inheritance, for example when instantiating FullPrintConfig and casting it to let's say HostConfig using dynamic_cast<>(), the HostConfig will still report all the aggregated keys of FullPrintConfig. Similarly, all the methods of ConfigBase (for example has(), apply(), equals(), serialize() etc) will work with the aggregated FullPrintConfig keys. I am not sure if this is what @alexrj has intended and whether this behavior has not some unintended consequences.

lordofhyphens, 2016-08-21:
So the patch added a constructor for FullPrintConfig that calls the child
constructors with a flag set so it doesn't actually do anything w/r/t
setting the defaults?

It then sets the defaults for everything at once at that level?

Just making sure I understand what was done here :)

lordofhyphens, 2016-08-21:
I will cherry pick it for a PR :)

bubnikv, 2016-08-21:
> It then sets the defaults for everything at once at that level?
> Just making sure I understand what was done here :)

You understand it very well.

Maybe you can find a more elegant approach, but this is simple enough.

I would personally love to get rid of the virtual inheritance. It took me
couple of hours to dig through the current code and  then to refresh my
knowledge of the virtual inheritance quirks.

lordofhyphens, 2016-08-21:
I think virtual inheritance is generally warranted in this case, given all
of the different configuration types, but I will review the current source
as it stands.

bubnikv, 2016-08-22:
Virtual inheritance is a complex tool, which all modern languages (at least
the languages I am aware of) avoid for a good reason. It is so complex,
that it is difficult to use correctly. Even multiple inheritance is
replaced by interfaces in java or c# and in most cases the multiple
inheritance is better replaced with object composition. To understand this
code I had to dust off the C++ documentation and I spent about an hour
studying virtual inheritance. I think object composition would work
sufficiently in the case of Slic3r configuration objects and it would make
the code easy to understand.

lordofhyphens, 2016-08-27:
Odd, this should've been auto-closed when the associated PR was merged.

closed 2016-08-27 by lordofhyphens