slic3r/Slic3r #3473: Make the boost detection more robust

bubnikv, 2016-08-21:
Slic3r is known to fail to detect boost library when compiling under Strawberry perl.

I tried to revamp the detection also with the goal of supporting Visual Studio compiler suite. Here is my attempt to it. @alexrj @lordofhyphens please inspect this.
https://github.com/prusa3d/Slic3r/commit/ab357c75a5746cf21fe7cb531ff849e9843f775c

I also added two new environment variables, BOOST_INCLUDEPATH and BOOST_LIBRARYPATH, names inspired by CMake. The BOOST_LIBRARYPATH is handy, if one has multiple stage directories, for example for 32/64bit builds, for debug/release builds, for Visual Studio/GCC builds etc.

The checkin also contains small number of changes required for Slic3r to compile with Visual Studio 2013 free compiler suite. Because the Visual Studio has a superb debugger, I believe it makes sense to support it.

lordofhyphens, 2016-08-21:
Yeah, the boost autodetection is fragile :(

I recall someone else working on the build scripts as well to clean them up.

alexrj, 2016-12-22:
This was merged, closing.

closed 2016-12-22 by alexrj