slic3r/Slic3r #3478: print preview on top instead of bottom of printer

lprevatte123, 2016-08-22:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-08-23:
Not enough information here. Please read/follow the instructions in italics above and edit your issue accordingly.

Otherwise, I will have to close the issue.

lordofhyphens, 2016-08-23:
Closing. Edit your post and I will reopen.

closed 2016-08-23 by lordofhyphens