slic3r/Slic3r #3479: Perimeter unglued them

greg60greg, 2016-08-23:
### Version
_Version of Slic3r used goes here_http://slic3r.org/

_Use `About->About Slic3r` for release versions_1.2.9

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 7

### Behavior
* _Describe the problem_ 

My scope are not glued together I think a recovery function between the perimeter must be included.
I had no problem on a standard printer , but I designed a very specific model and perimeters arises each other 's sides by little sticky ( or to a collection function necesary I think)

* _Steps needed to reproduce the problem_

Each layer in the vertical direction (except in the first layer strangely )

  * _If this is a command-line slicing issue, include the options used_

* _Expected Results_ 
Perimeters that slightly overlaps them

* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_  Yes

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-08-23:
I don't understand what you are asking for.  Please add pictures/ screenshots and a way to reproduce the behavior  (configuration +test STL ).

lordofhyphens, 2016-08-23:
Also, look at the perimeters/infill overlap setting.

greg60greg, 2016-08-23:
Hello,

My scope are not glued together I think a recovery function between the
perimeter must be included.
I had no problem on a standard printer , but I designed a very specific
model and perimeters arises each other 's sides by little sticky ( or to a
collection function necesary I think).

My problem does not occur between the infill and perimeter but between
the outer perimeters and I value that does not change the first
perimeter is well glued to the infill .

And I have no problem in extrusion or development extruder I spend all
scrutinized.

*Expected Results*
Perimeters that slightly overlaps them

Thanks you for your reponse (sorry for my english i am french)

greg60greg, 2016-08-23:
[image]
[image]

lordofhyphens, 2016-08-23:
Please include a 3d model you are seeing this problem on and your slic3r configuration ("File->Export Configuration"). If I can reproduce the problem I can better understand it. You should be able to drag/drop zip files onto the edit area to upload.

Since perimeters are marked when 'verbose' gcode is turned on, one workaround you could try is a short post-processing script to increase extrusion amount slightly for just perimeter moves.

I'm not sure what you mean by "recovery". 

Could you also give it a try with this version: https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-master--566bbbd.zip

greg60greg, 2016-08-23:
This are not relative to the 3D model all my files have the same problem.
I want to try any worries. I note that with advanced in the perimeters 200% I have the same thing has the first bedroom which pretty hadère. I'm going to try.
No worries to try with the other version. I do not see it setting you speak?.

My problem is really in the software at the overlap of the perimeters.

After I print of PET is that it can play?.

[config.zip](https://github.com/alexrj/Slic3r/files/432903/config.zip)

lordofhyphens, 2016-08-23:
The link I posted is a copy of the current development version, which will eventually become the next release.

Also: Export config, not export config bundle :)
[image]

A 3d model that shows the issue saves the time/confusion of "well, I don't see this problem".  Because I haven't seen this problem on my printers with anything I've done with Slic3r.

It's possible that your first layer looks fine because Z=0 may be set slightly low (squishing it), and such upper layers could be slightly underextruded.

greg60greg, 2016-08-23:
Yes but what is strange is that the problem only occurs on the perimeter. Even the filling is done properly and more that I checked my recovery has multiple extruder and the flow is good.
I will recheck still a lot of points and coming back to you.

For example if I would print with a single perimeter that I would not worry. But the interest to the outline several times is not explained and is indispensable.

Thank you for your help
[config.zip](https://github.com/alexrj/Slic3r/files/433092/config.zip)

bubnikv, 2016-08-23:
I suppose @greg60grep extrudes a too thin line? I suppose you need to
calibrate your extruder?
How about the solid infill? Does it fill the layer completely?
Why the perimeters do not bond? Is the extrudate too cold? Is the already
printed piece too cold? Under extrusion?
Is the problem specific to PET, or do you have the same problem with PLA?

greg60greg, 2016-08-23:
The extruder is set I have a 0.5mm nozzle and my flow is checked and I have a difference of 0.3% is nothing.
Everything goes well throughout the finished piece is clean but if I put my nail comes off the perimeters of each other.
So I go to a room that was perfect a piece with the perimeters to peel apart so big quality concerns.
I think the concerns would be identical with the PLA but I did not even try I will check all of my values to my self sensor bed just now but I do not think it come from but you never know .

greg60greg, 2016-08-23:
he problem disappears when I advanced in perimeter included 100% tomorrow I try 200% that it is the default value of this value when the box is 0?.

Thank you

lordofhyphens, 2016-08-23:
100% in the perimeter width box means that the perimeter width is the same as the layer height. This is probably not what you want (as the layer height is generally less than your nozzle width, which is not a good thing).
[image]

You could see what happens if you set perimeters and external perimeters to some fixed value like 0.6 or 0.75.

100% in the infill/overlap box means that it printed the infill and perimeter in the same place.
[image]

lordofhyphens, 2016-08-23:
The default value is shown in the output .gcode file at the top.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

greg60greg, 2016-08-25:
Hello thé problem is solved, he is goes of one erreur in thé autobed level of Marlin.
Sorry ans thanks.

closed 2016-08-25 by lordofhyphens