slic3r/Slic3r #3480: Issue/Feature request: Auto split before generating GCode

Iffoffo, 2016-08-24:
### Version
1.2.9

### Operating system type + version
Win 7 64bit

### Issue
* I use Repetier 1.6.2 and I'd like to do "Sequential Printing" but when Repetier sends objects to Slic3r, Slic3r see them as only one STL (only one object) so it can not complete individual objects

### New Feature
* I'd like to have a checkbox "Auto-split before generating GCode = Y/N". In this way I think that we could solve this issue

Many thanks

lordofhyphens, 2016-08-24:
I will check the CLI interface to see if it supports multiple input files.
I think split on load through the CLI is probably generally useful though
(if it was just for RepetierHost it probably wouldn't get in, we are
generally disinclined to make changes solely to support closed source
programs).

alexrj, 2016-11-24:
Repetier-Host should export an AMF file instead of a STL. AMF supports multiple objects. Also, slicing will be faster when you put multiple copies of an object on plate.

closed 2016-11-24 by alexrj

bubnikv, 2016-11-24:
Repetier indeed generates AMF. Only AMF import is many orders of magnitude
slower than STL import. I am working on C++ port of the AMF import.