slic3r/Slic3r #3482: Feature request: Change extruder (or mix) at infill

kyrreaa, 2016-08-24:
It would be very useful if there was a way to instruct Slic3r to do the tool change when it is done with the perimiter of the previous extruder instead of waiting till it wants to start on perimiter of the new extruder.
Alternately if "Infill before perimeters"-option is enabled it can switch when starting on the new extruders infill as this could be invisible and used to hide the color change.

Obviously not a good idea to enable this for water soluble support etc, but for single orifice hotends with color changing or blending it would allow priming to be done while doing something useful instead of prime towers or external purge/dumping.

lordofhyphens, 2016-08-24:
Was pretty sure purging/toolchanges were why we have skirts.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

kyrreaa, 2016-08-24:
Skirts?
Where to begin...

First off with a single nozzle, it does not drip so no reason to use ooze guarding skirt for that.

Secondly, on the hotends with a very narrow space between nozzles the skirt does not help much on ooze as both nozzles are within the skirt.

Thirdly the skirt blocks for the cooling fan keeping more of the heat in.  This is good if you need that but ruins the printspeed otherwise as you have to slow down for overhangs to cool properly. When you print at 140mm/s you quickly realize how slowly pla cools...

Oh, and you cannot deposit 7 color changes worth of material in a skirt.
Try to dump it outside the skirt and you need something to even that out too.

Here I use custom toolchange code with Cura:
https://www.youtube.com/watch?v=tdA65HTppWA

lordofhyphens, 2016-08-24:
In terms of options (not sure how to do it yet), what about a priority list of legal places to do purges?

kyrreaa, 2016-08-24:
A priority list would be awsome.
I was just thinking that simply moving the layer change in some cases was a very easy thing to do in the code vs in a postprocessor.

kyrreaa, 2016-08-24:
Left pillar is printed with reverse order and interface shells.
This allows a larger area for infill when starting the new piece to maximize the clearing of old color.
The right pillar is normal order and no interface shells.

It is most noticable on the yellow and green that 1-2 layers are tinted by old color on right pillar.
There was no purging or similar done on these prints, just reversing the order using the infill before perimiter option. My idea is simply not to do this for the previous layer/object, but instead use that parts infill. Then continue onto the new parts infill. Ofcource, if multiple color changes are to occur, then I guess the prev parts infill has already been used to purge...

[image]

rob-miller, 2017-06-11:
see also https://somei3deas.wordpress.com/2017/01/06/multi-colour-printing-without-wipe-or-prime-towers/ and https://somei3deas.wordpress.com/2017/01/24/latest-update-to-blog-6th-jan-multi-colour-printing-without-wipe-and-prime-towers/ where the author post-processes slic3r output to move the tool change to infill areas.