slic3r/Slic3r #3483: Bridging speed used on top of dense infill rather than solid-infill speed

MortarArt, 2016-08-25:
### Version
All

### Operating system type + version
All

### Behavior
* Bridging speed over dense infill is too slow
* Example settings:
  * Bridges speed 30mm/s
  * Solid infill speed 140mm/s
  * Fill density 80%
  * Fill pattern Rectilinear
* Should allow for printing of solid infill layers at 140mm/s
* Actually results in solid infill layers at 30mm/s

FlameRunner, 2016-08-25:
It's not a bug, it's a feature. Not only bridge speed is used, but also bridge flow. That's a quite good way to address problem with solid infill lines breaking over the low density sparse infill.

But for high density infill, it doesn't really make sense and can lead to overextrusion.

I have thought about that problem some time ago and I believe that the solution would be to add 'use bridge speed and flow for first layer of solid infill over sparse infill' checkbox in 'infill' section.

lordofhyphens, 2016-08-25:
See pull request.

closed 2016-08-25 by lordofhyphens

bubnikv, 2016-09-26:
It would be great, if the config wizard was also extended to support this config.