slic3r/Slic3r #3484: GNU Affero General Public License message

debakantasandha, 2016-08-26:
While running Slic3r from CMD Command prompt I'm getting GNU Affero General Public License message. Though it is producing the gcode file its contents are not as complete as it is produced through the GUI.
Can you please explain the  steps to overcome this issue?

Thanks & Regards,
Debakanta

lordofhyphens, 2016-08-27:
I really have no idea what's going on here. Please include a copy of the messages you see and describe more fully what is going on. The Gcode generated on both the CLI and GUI (with test STL file) and exported config.ini are going to be necessary as well. 

When you wrote this issue you deleted a lot of questions/messages. Those are generally not optional questions to skip if you want any kind of meaningful help. Please review the contributing guidelines for more information.

What I can tell you is that unless you use --load to load a configuration, the --default-- configuration gets used, which is probably not what you want.

alexrj, 2016-12-16:
@debakantasandha, can you please tell what Slic3r version are you using? Where did you download it from? What operating system? How are you invoking it?

alexrj, 2017-02-08:
No answer from @debakantasandha, closing

closed 2017-02-08 by alexrj