slic3r/Slic3r #3485: 'Pillars' Infill columns may not be reliably printable

lordofhyphens, 2016-08-27:
### Version
_Version of Slic3r used goes here_
1.2.9, 1.3.0-dev
### Operating system type + version
All

### Behavior
* The thin towers produced by Slic3r may not be reliably printable with some printers.

[image]

_Is this a new feature request?_
Yes

### Suggested solution
The two primary options for pillar infill, spacing and size, are set as constants in SupportMaterial.pm.
While it is certainly feasible to make those options in the configuration file, need to solve some UI questions first to avoid cluttering the UI.

#### STL/Config (.ZIP) where problem occurs
[issue.zip](https://github.com/alexrj/Slic3r/files/440507/issue.zip)
Note: rotate the stl 180degrees around the X axis

lordofhyphens, 2016-08-27:
I've got an experimental (just looking at adding config options, not dealing with the UI) branch at: https://github.com/lordofhyphens/Slic3r/tree/config-pillar.

I've left the default options to what the constants were, so existing tests pass. I do need to come up with a few tests.

lordofhyphens, 2016-08-27:
@alexrj any comments with regards to your preferences as to UI, etc?

Lord2Vader, 2016-08-30:
Have you tried using a wider line width for the support? Works out for me most of the time. Although yes these pillars are very veeery fragile.

lordofhyphens, 2016-08-30:
I don't usually have problems, but I have heard them from others.

Since the parameters are already there, it is pretty straightforward to add
them to config (probably behind an optional menu that only is available
after setting that type.

DrLex0, 2016-09-03:
The biggest problem I see with the pillars is lack of footprint. A tall narrow pillar becomes very easy to knock over unless bed adhesion is insanely good. Even if it is not knocked over, when a narrow pillar becomes tall enough, it becomes too flexible and wobbles around while things are printed on top of it.

Pillars should be the wider, the taller they are. At least their footprint should be, the top end may be narrower.

[image]
This was my first print with tall narrow pillars (and also the last, now I construct my own support material for prints like these). It is a bit of a miracle that it ended up usable. It cracked from opening the door and letting cool air in while I prevented the mess from sticking to the nozzles and further ruining things. I filled the cracks and reconstructed the missing overhang with exoxy glue, and it was a usable spool holder.

lordofhyphens, 2016-09-03:
I have an experimental fork available on bintray.com that permits changing
the pillar size.

lordofhyphens, 2016-09-03:
@DrLex0 https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches/1.3.0-config-pillar-176

mrjohnc, 2016-10-04:
Would it be possible to: 

1. Add little 'bridges' between the columns to help them support each other and spread the forces
2. Flare the bottoms of the pillars when they are touching the buildplate
3. Add flying butresses, this may be particularly helpful for towers that are not fixed to the bed but a part of the model and so cannot have larger contact to the bed https://en.wikipedia.org/wiki/Flying_buttress

mavu, 2016-10-05:
Another idea would be to make the pillars round. this would make it easier on the structure while printing due to avoiding harsh left-right movement in favor of "softer" circular movements.

bubnikv, 2017-02-02:
I find the geometry of the pillars to be generated unreliably by the current Slic3r. Some of the supported areas have a very thin or non-existent pillar generated, leading to a spoiled print.

I find the zig-zag support with no sheath as implemented in the latest prusa3d form to be much more robust. When printed with a thin extrusion line, the support likely does not use much more material than the solid towers.

pierbiag, 2017-09-18:
I agree with mrjohnc on the bridging between them: it may either be a full solid layer every "N" layers of pillar, or, even better: the pillar flaring out every "N" layers, connecting with the others and then flaring back in, in a diamond shape.

nylen, 2018-03-01:
How about simply disabling retraction *in between* support islands, not only *within* a single support island?

It looks like this could be done fairly easily here:  https://github.com/alexrj/Slic3r/blob/dc9d7ed7/xs/src/libslic3r/GCode.cpp#L682-L688

I can give it a try if this is something you'd be interested in.

mrjohnc, 2018-03-01:
@nylen sure, give it a try :)

nylen, 2018-03-01:
Submitted #4327.  It needs some work but it seems like it helps with this issue on my printer.