slic3r/Slic3r #3486: Slic3r crashes on 1.2.9, 1.2.8, 1.2.7 but NOT 1.2.2 on OSX

DouglasPearless, 2016-08-27:
Attached are my config and a STL file that always crashes on 1.2.9, 1.2.8, 1.2.7 but NOT 1.2.2 on OSX.
Steps:
(1) Load config
(2) Load STL
Slic3r begins to generate perimeters, infill and crashes
[Archive.zip](https://github.com/alexrj/Slic3r/files/440514/Archive.zip)

lordofhyphens, 2016-08-27:
Could you run this from the command line and include any messages generated by slic3r?

Also, 100% fill density is probably counterproductive :)

DouglasPearless, 2016-08-27:
Ok, but I forgot to mention I also tried 60% infil.

lordofhyphens, 2016-08-27:
I haven't seen a crash on Linux with 1.3.0-dev (yet), but it is taking its sweet time generating the infill. If I can't reproduce the crash on Linux we'll have to find someone willing to run the problem down on OSX.

I'd also suggest trying Grid infill at 50% or honeycomb at 30%.

lordofhyphens, 2016-08-27:
This is in addition to trying to get any error messages that may be output on crash. You'll need to run Slic3r from the command line to get this kind of information.

DouglasPearless, 2016-08-27:
From the command line I get:

/Applications/Slic3r.app/Contents/MacOS/slic3r 3D0001e\ -\ Delta\ Motor\ Assembly\ -\ Dual\ 608\ Bearing\ Block.stl --load 3KU_config_22.ini 
=> Processing triangulated mesh
=> Generating perimeters
=> Preparing infill
Segmentation fault: 11

I didn’t see a verbose mode to generate more diagnostic information, here is the crash report:

Process:               slic3r [51652]
Path:                  /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:            slic3r
Version:               Slic3r 1.2.9.99 (1.2.9.99)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           slic3r [51652]
User ID:               501

Date/Time:             2016-08-27 18:08:46.211 +1200
OS Version:            Mac OS X 10.11.6 (15G31)
Report Version:        11
Anonymous UUID:        AA2210E0-A9FF-63EF-6194-2D409DB01287

Sleep/Wake UUID:       842743C1-D204-410B-AC41-8468E524F5BF

Time Awake Since Boot: 260000 seconds
Time Since Wake:       110 seconds

System Integrity Protection: enabled

Crashed Thread:        5  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
objc[51652]: autorelease pool page 0x436e000 corrupted
  magic     0x00000000 0x00000000 0xa1a1a1a1 0x4f545541
  should be 0xa1a1a1a1 0x4f545541 0x454c4552 0x21455341
  pthread   0x21455341
  should be 0xb0115000
 

Thread 0:: Dispatch queue: com.apple.main-thread
0   libperl.dylib                 	0x000e70ac Perl_pp_entersub + 1276
1   libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
2   libperl.dylib                 	0x00074ca5 Perl_call_sv + 725
3   Wx.bundle                     	0x07d75e67 wxPliEventCallback::Handler(wxEvent&) + 1111
4   libwx_baseu-3.0.0.0.0.dylib   	0x0882dd9f wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const + 95
5   libwx_baseu-3.0.0.0.0.dylib   	0x089757a8 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 104
6   libwx_baseu-3.0.0.0.0.dylib   	0x08975f98 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 88
7   libwx_baseu-3.0.0.0.0.dylib   	0x08976904 wxEvtHandler::TryHereOnly(wxEvent&) + 52
8   libwx_baseu-3.0.0.0.0.dylib   	0x089769ac wxEvtHandler::ProcessEventLocally(wxEvent&) + 60
9   libwx_baseu-3.0.0.0.0.dylib   	0x08976a21 wxEvtHandler::ProcessEvent(wxEvent&) + 81
10  libwx_baseu-3.0.0.0.0.dylib   	0x089763d7 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 39
11  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x08412bfb wxWindowBase::HandleWindowEvent(wxEvent&) const + 27
12  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x084131fc wxWindowBase::SendIdleEvents(wxIdleEvent&) + 156
13  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x084131c7 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 103
14  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x084131c7 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 103
15  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x084131c7 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 103
16  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x084131c7 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 103
17  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x084131c7 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 103
18  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x082f16d9 wxAppBase::ProcessIdle() + 153
19  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x0824e7f2 wxApp::ProcessIdle() + 34
20  libwx_baseu-3.0.0.0.0.dylib   	0x0886e926 wxEventLoopBase::ProcessIdle() + 38
21  libwx_baseu-3.0.0.0.0.dylib   	0x0893fa98 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 120
22  com.apple.CoreFoundation      	0x96dbf90e __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 30
23  com.apple.CoreFoundation      	0x96dbf86e __CFRunLoopDoObservers + 398
[... 399 lines cut ...]
0x9d7f0000 - 0x9d8e6fff  com.apple.QuickLookUIFramework (5.0 - 696.7) <33B2BB97-1632-39D7-9DF0-CD0EE7F00625> /System/Library/Frameworks/Quartz.framework/Versions/A/Frameworks/QuickLookUI.framework/Versions/A/QuickLookUI
0x9d8e7000 - 0x9d90dff7  com.apple.quartzfilters (1.10.0 - 1.10.0) <C742B6D3-6E5E-362A-86BD-0143A79D8225> /System/Library/Frameworks/Quartz.framework/Versions/A/Frameworks/QuartzFilters.framework/Versions/A/QuartzFilters
0x9d90e000 - 0x9d9b5fff  com.apple.Metadata (10.7.0 - 972.34) <ADC9277D-F155-32E0-B6D8-6D2DB90FFB84> /System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/Metadata.framework/Versions/A/Metadata
0x9d9b6000 - 0x9d9ccfe7  libcompression.dylib (28) <B7A86671-648B-35B3-9A61-5B36D074689C> /usr/lib/libcompression.dylib
0x9d9cd000 - 0x9d9d8fff  com.apple.NetAuth (6.0 - 6.0) <5BDBB960-B818-3B5A-9F0A-345DFB727F76> /System/Library/PrivateFrameworks/NetAuth.framework/Versions/A/NetAuth
0x9d9d9000 - 0x9de0bffb  com.apple.vision.FaceCore (3.3.1 - 3.3.1) <82AA6556-43A3-3EF5-BE6E-2653207F4619> /System/Library/PrivateFrameworks/FaceCore.framework/Versions/A/FaceCore
0x9de0c000 - 0x9de61ff7  com.apple.htmlrendering (77 - 1.1.4) <5EFC54AC-7FD8-3ADA-97A8-345DD71A9D85> /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HTMLRendering.framework/Versions/A/HTMLRendering
0x9de62000 - 0x9dec9fff  libcorecrypto.dylib (335.50.1) <66992865-A8BF-33C4-8C7D-A5EA7A13E90D> /usr/lib/system/libcorecrypto.dylib
[... cut ...]
0x9e9bf000 - 0x9eb7bff7  com.apple.avfoundation (2.0 - 1046.9.12) <3E0D2C68-5A0D-3060-9BF1-4CC0747DAA82> /System/Library/Frameworks/AVFoundation.framework/Versions/A/AVFoundation
0x9eb7c000 - 0x9eb86fff  libsystem_notify.dylib (150.40.1) <2289C342-4E02-3DE8-9160-A2A4E288B2DD> /usr/lib/system/libsystem_notify.dylib
0x9eb87000 - 0x9eb93ff7  libcsfde.dylib (517.50.1) <FE525CD1-6237-37F9-8E62-F92AD012587B> /usr/lib/libcsfde.dylib
0x9eb94000 - 0x9ebd7fff  libGLU.dylib (12.1) <A30631F5-B0F9-38E0-8881-96270645B1AF> /System/Library/Frameworks/OpenGL.framework/Versions/A/Lib

lordofhyphens, 2016-08-27:
I ran the model through the online netfabb and it was fine on Linux 1.3.0-dev. I'm not sure that 150% overlap does what you want it to, but w/e.

lordofhyphens, 2016-08-27:
[fixed_stl.zip](https://github.com/alexrj/Slic3r/files/440740/fixed_stl.zip)

DouglasPearless, 2016-08-28:
I tried the fixed sol and notices the top 4 screw hols were partially blocked, but again I got a crash:

Process:               slic3r [61083]
Path:                  /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:            slic3r
Version:               Slic3r 1.2.9.99 (1.2.9.99)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           slic3r [61083]
User ID:               501

Date/Time:             2016-08-28 17:47:12.785 +1200
OS Version:            Mac OS X 10.11.6 (15G31)
Report Version:        11
Anonymous UUID:        AA2210E0-A9FF-63EF-6194-2D409DB01287

Sleep/Wake UUID:       E0AF78B4-A487-41B7-B82C-0A1EE62B4242

Time Awake Since Boot: 280000 seconds
Time Since Wake:       470 seconds

System Integrity Protection: enabled

Crashed Thread:        4

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000394
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x394:
--> 
    __TEXT                 0000000000001000-0000000000038000 [  220K] r-x/rwx SM=COW  /Applications/Slic3r.app/Contents/MacOS/slic3r

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreGraphics        	0x90ec52c4 blt_pattern_blend_XXXX32 + 1078
1   com.apple.CoreGraphics        	0x90e7e762 argb32_mark + 20855
2   com.apple.CoreGraphics        	0x90e98066 argb32_image + 2422
3   libRIP.A.dylib                	0x9b3fc101 ripd_Mark + 278
4   libRIP.A.dylib                	0x9b3fa9c4 RIPLayerBltImage + 1150
5   libRIP.A.dylib                	0x9b3fa2ec ripc_RenderImage + 175
6   libRIP.A.dylib                	0x9b403627 ripc_DrawImages + 7318
7   com.apple.CoreGraphics        	0x90ebff2b CGContextDrawImages + 242
8   com.apple.coreui              	0x9aa97b5f DrawNinePartImageWithOperation + 4997
9   com.apple.coreui              	0x9aa967c0 DrawNinePartElementFromRenditionWithOperation + 531
10  com.apple.coreui              	0x9aa9e249 -[CUIThemeFacet _drawSpecificRenditionKey:rendition:inFrame:context:alpha:operation:isFocused:isFlipped:] + 762
11  com.apple.coreui              	0x9aa9de64 -[CUIThemeFacet _drawSpecificRenditionKey:inFrame:context:isFocused:isFlipped:] + 168
12  com.apple.coreui              	0x9aa9bff8 -[CUIThemeFacet drawInFrame:isFocused:context:] + 135
13  com.apple.coreui              	0x9aaa8a80 CUICoreThemeRenderer::DrawSimple(CUIRenditionKey*, CUIDescriptor const*, signed char) + 506
14  com.apple.coreui              	0x9aaa8e87 CUICoreThemeRenderer::DrawTabPane(CUIDescriptor const*) + 155
15  com.apple.coreui              	0x9aa31c7a CUIRenderer::Draw(CGRect, CGContext*, __CFDictionary const*, __CFDictionary const**) + 2462
16  com.apple.coreui              	0x9aa344d9 CUIDraw + 209
17  com.apple.AppKit              	0x9c9f0bd9 __44-[NSAppearance _drawInRect:context:options:]_block_invoke + 65
18  com.apple.AppKit              	0x9c841b69 -[NSAppearance _callCoreUIWithBlock:] + 25
19  com.apple.AppKit              	0x9c841af0 -[NSCompositeAppearance _callCoreUIWithBlock:] + 163
20  com.apple.AppKit              	0x9c9f0b91 -[NSAppearance _drawInRect:context:options:] + 127
21  com.apple.AppKit              	0x9cbc1b69 -[NSTabView _coreUIDrawBezelInRect:withClip:flipped:] + 297
22  com.apple.AppKit              	0x9cbc199c -[NSTabView _drawThemeBezelBorder:inBounds:clipRect:] + 496
23  com.apple.AppKit              	0x9cbc143d -[NSTabView _drawBezelBorder:inRect:] + 137
24  com.apple.AppKit              	0x9cb996dd -[NSTabView _drawBorder:inRect:] + 132
25  com.apple.AppKit              	0x9cb9946d -[NSTabView drawRect:] + 283
26  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x073e2d4f wxWidgetCocoaImpl::drawRect(void*, NSView*, void*) + 847
27  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x073e3420 wxOSX_drawRect(NSView*, objc_selector*, _NSRect) + 368
28  com.apple.AppKit              	0x9c9fee78 -[NSView _drawRect:clip:] + 3700
29  com.apple.AppKit              	0x9c9fd7a7 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 5263
30  com.apple.AppKit              	0x9c9fd17a -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 3682
31  com.apple.AppKit              	0x9c9fd17a -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 3682
32  com.apple.AppKit              	0x9c9fbe9a -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 313
33  com.apple.AppKit              	0x9c9fa53e -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2622
34  com.apple.AppKit              	0x9c9f562f -[NSView displayIfNeeded] + 1687
35  com.apple.AppKit              	0x9c9f4f33 -[NSWindow displayIfNeeded] + 348
[... 482 lines cut ...]
shared memory                     16.4M       11 
===========                     =======  ======= 
TOTAL                            949.2M     1448 
TOTAL, minus reserved VM space   949.2M     1448 

Model: MacBookPro8,2, BootROM MBP81.0047.B2C, 4 processors, Intel Core i7, 2 GHz, 8 GB, SMC 1.69f4
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Graphics: AMD Radeon HD 6490M, AMD Radeon HD 6490M, PCIe, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
[... cut ...]
Bluetooth: Version 4.4.6f1 17910, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Crucial_CT512M550SSD1, 512.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

Astroman, 2016-09-06:
Still crashing pretty regularly on my MacBook Pro running latest OS. Seems to crash more with complex objects, especially when I try to do something while it is slicing. Otherwise, it's working great.

lordofhyphens, 2016-09-06:
I don't have a Mac, so I can't debug this any further.

@bubnikv or @alexrj may have some insights.

lordofhyphens, 2016-09-20:
@DouglasPearless @Astroman Turn off background processing, see if you can reproduce crash.

DouglasPearless, 2016-09-20:
OK, tried that, it still crashes on "Preparing Infill":

Process:               slic3r [53252]
Path:                  /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:            slic3r
Version:               Slic3r 1.2.9.99 (1.2.9.99)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           slic3r [53252]
User ID:               501

Date/Time:             2016-09-21 11:19:31.408 +1200
OS Version:            Mac OS X 10.11.6 (15G1004)
Report Version:        11
Anonymous UUID:        AA2210E0-A9FF-63EF-6194-2D409DB01287

Sleep/Wake UUID:       F9DD709A-DE16-4317-9CD4-0A23EA433BCB

Time Awake Since Boot: 160000 seconds
Time Since Wake:       9100 seconds

System Integrity Protection: enabled

Crashed Thread:        2  com.apple.NSEventThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000020a6ebfe
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x20a6ebfe:
    Process Corpse Info    0000000020600000-0000000020800000 [ 2048K] rw-/rwx SM=COW  
--> 
    __TEXT                 000000003e000000-000000003e042000 [  264K] r-x/rwx SM=COW  /System/Library/Frameworks/GLUT.framework/Versions/A/GLUT

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x9b23d49a mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x9b23c884 mach_msg + 68
2   com.apple.CoreFoundation      	0x97ecfdc6 __CFRunLoopServiceMachPort + 214
3   com.apple.CoreFoundation      	0x97ecf1d1 __CFRunLoopRun + 1521
4   com.apple.CoreFoundation      	0x97ece976 CFRunLoopRunSpecific + 390
5   com.apple.CoreFoundation      	0x97ece7db CFRunLoopRunInMode + 123
6   com.apple.HIToolbox           	0x91f982f1 RunCurrentEventLoopInMode + 267
7   com.apple.HIToolbox           	0x91f980f3 ReceiveNextEventCommon + 503
8   com.apple.HIToolbox           	0x91f97eec _BlockUntilNextEventMatchingListInModeWithFilter + 99
9   com.apple.AppKit              	0x94fee2e2 _DPSNextEvent + 1053
10  com.apple.AppKit              	0x94fed85b -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1057
11  com.apple.AppKit              	0x94fed432 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 121
12  com.apple.AppKit              	0x94fe0b47 -[NSApplication run] + 1063
13  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x07ebc685 wxGUIEventLoop::OSXDoRun() + 277
14  libwx_baseu-3.0.0.0.0.dylib   	0x085397c6 wxCFEventLoop::DoRun() + 38
15  libwx_baseu-3.0.0.0.0.dylib   	0x084689ba wxEventLoopBase::Run() + 74
16  libwx_baseu-3.0.0.0.0.dylib   	0x08428a3f wxAppConsoleBase::MainLoop() + 95
17  Wx.bundle                     	0x07875be2 wxPliApp::MainLoop() + 50
18  Wx.bundle                     	0x0781a8b1 XS_Wx__App_MainLoop(interpreter*, cv*) + 129
19  libperl.dylib                 	0x000e74a6 Perl_pp_entersub + 2294
20  libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
21  libperl.dylib                 	0x0007542e perl_run + 366
22  slic3r                        	0x00006d2b 0x1000 + 23851
23  slic3r                        	0x000045a5 0x1000 + 13733

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x9b2457fa kevent_qos + 10
1   libdispatch.dylib             	0x9b0dc7ea _dispatch_mgr_invoke + 234
2   libdispatch.dylib             	0x9b0dc3be _dispatch_mgr_thread + 52

Thread 2 Crashed:: com.apple.NSEventThread
0   libsystem_malloc.dylib        	0x904fcf65 small_malloc_from_free_list + 988
1   libsystem_malloc.dylib        	0x904f9ef3 szone_malloc_should_clear + 1229
2   libsystem_malloc.dylib        	0x904f9a1f szone_malloc + 24
3   libsystem_malloc.dylib        	0x904f99c7 malloc_zone_malloc + 75
[... 445 lines cut ...]
shared memory                     16.4M       10 
===========                     =======  ======= 
TOTAL                            878.1M     1531 
TOTAL, minus reserved VM space   878.1M     1531 

Model: MacBookPro8,2, BootROM MBP81.0047.B2C, 4 processors, Intel Core i7, 2 GHz, 8 GB, SMC 1.69f4
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Graphics: AMD Radeon HD 6490M, AMD Radeon HD 6490M, PCIe, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.4.6f1 17910, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Crucial_CT512M550SSD1, 512.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

bubnikv, 2016-09-23:
Did you try the Prusa3D builds? They use the fresh GUI libraries.
https://github.com/prusa3d/Slic3r/releases

Astroman, 2016-09-24:
The Prusa3D build looks good so far. It hasn't crashed and seems a little faster as well. Thanks!

alexrj, 2017-03-13:
@DouglasPearless, @Astroman, we now have fresh automated builds available at
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view

They are made on a recent OSX machine and I expect them to be very stable. Please reopen this if you have problems with those builds. Thank you for your report.

closed 2017-03-13 by alexrj