slic3r/Slic3r #3488: compiling hangs on "Building and testing Alien-wxWidgets-0.67"

phizz166, 2016-08-28:
Sorry to use the issue tracker like this, but I couldn't find anyone else with the same issue or an appropriate place to discuss it. If there's a better spot for these kind of tech support questions, please let me know.

Trying to compile the latest version from source (Win 10, Strawberry Perl). I can compile the CLI version fine. Trying to compile the GUI, I get up to this point:

```
==> Found dependencies: Alien::wxWidgets
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz ... OK
Configuring Alien-wxWidgets-0.67 ... OK
Building and testing Alien-wxWidgets-0.67 ...
```
and it just hangs there. Half an hour so far. Killing the process and retrying, or using --force, doesn't do anything. Any ideas?

lordofhyphens, 2016-08-28:
Odd, as I used the build instructions from the wiki on my Win10 laptop and it didn't hang there, and our build server is Win8.1 and it repeatedly builds wxWidgets.

@bubnikv any ideas?

lordofhyphens, 2016-08-28:
As an aside: 
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-177#files has a built Win32 version with a batch file to run it. That's an automatic build as things are committed. 
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches for different branches (proposed changes to slic3r).

phizz166, 2016-08-28:
So literally 30 seconds after I posted this, it finished that step and kept on going. But now it's hanging on Building and testing Wx-0.9928. There's a little gray window popped up, title bar "X", and it's (not responding). 

I guess the compilation is still going, but any reason that it would be getting stuck here so badly? My laptop has a 2.5GHz i5 and 4GB of RAM, and the Perl interpreter is using 100% of one core. The command-line version compiled without any obvious delays.

lordofhyphens, 2016-08-28:
It is probably still building and testing wx, you can probably close that
window.

lordofhyphens, 2016-08-28:
That is basically all library stuff which we don't have any real control
over :(.

phizz166, 2016-08-28:
Force-closed that X window, and it reported that that particular test failed. Will that have an effect?

Now it's moved on to
...
==> Found dependencies: Wx::build::MakeMaker
--> Working on Wx::build::MakeMaker
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
Configuring Wx-0.9928 ... OK
Building and testing Wx-0.9928 ...
...

and is still stuck there.

Oh well, at least it's going. I will try out the binaries you've got, but I wanted to start making some changes to the code myself.

lordofhyphens, 2016-08-30:
Oh wxperl. Nobody likes you.

Since the build script only basically runs cpanm, you can try to build it from the command line yourself and look through the build log.

The usual suspect for me and figuring out what is going on with wxperl on windows is to make sure that your PATH is only pointing at one set of compiler and friends. Having binutils (make, etc) from cygwin installed and MINGW's (via MSYS) at the same time confuses the system mightily. I ran into a similar issue on the build server and basically had to delete an awful lot of stuff to get it to play nice. I think Visual Studio's tools also can get pulled in by accident if they're in the PATH. 

I've updated the [wiki](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows) with instructions helpfully provided by @bubnikv for building with Visual Studio 2013 (along with some patches he wrote for libraries). I hope that helps.

Brandano, 2016-09-12:
I had the exact same issue. Eventually I "fixed" it by manually running "mcpan --install Wx --force" from the strawberry perl command prompt, and forcing close the window titled "X". Slic3r then compiles fine, and seems to run correctly.

Brandano, 2016-09-12:
oops, sorry, that should be "cpanm  --install Wx --force"

bubnikv, 2016-09-13:
There is a single test case in the Perl::Wx package I believe, which hangs on Windows, when building for 64bit. I did not go any further into this, but I did not see any problems with the Wx installation. I am doing the same -  I am forcing the install manually.

Brandano, 2016-09-13:
I can confirm that this happens on 64 bit windows 10, compiling with mingw, but I guess that's a bug that should be probably reported upstream.

bubnikv, 2016-09-13:
You can try that, but my impression is the wxperl project is essentially abandoned. You will be better off, if you try to fix the problem on your own.

MartinMajewski, 2016-11-02:
Wx ist a pain in the back... I have somehow managed to build it wit cpan, but cpanm does not like it and building Slic3r with the gui flag never succeeded on my machine...

bubnikv, 2016-11-02:
One of the Wx test cases hangs on 64bit windows. If you force install Wx,
it will work.

MartinMajewski, 2016-11-02:
Nope, no chance... Forcing to install it with cpanm just tells me to use --force the next time... As said, it is not a problem to get it compile with cpan, but as Slic3r uses cpanm I am out of luck and I hate perl too much to investigate how to maybe use the cpan Wx build with cpanm...

bubnikv, 2016-11-02:
I am not in love with Perl either.

The cpan gives you a directory with the error log. THat is the directory it
compiled in. You need to close cpan, enter that directory and run some form
of make,
maybe just make install
or perl Makefile.PM install

You will surely find hints on how to compile Wx::Perl in my writeup on how
to compile with Visual Studio. Many of the steps apply to Strawberry as
well.
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows
section
6) Installing the GUI perl modules by Slic3r

By the way, the writeup is worth of one man week.

Vojtech

MartinMajewski, 2016-11-02:
cpan compiles Wx fine!
It is cpanm that is not able to.
I use perlbrew and I am on Mac, but surely the writeup is useful for Windows users.

My current status on this is: I use the Prusa Research Slic3r builds, as they get updates more frequently than the vanilla Slic3r builds. And if Slic3r does not suite my needs because of a bug or something similar (like the weak support material generation) I use Simplify3D. It is a shame, that such a great project like Slic3r is written in such an inconvenient nerd-on-principle-language.

No hard feelings!

bubnikv, 2016-11-02:
> cpan compiles Wx fine!
> It is cpanm that is not able to.

Then why don't you just install Wx with cpan, it shall work then.
Sorry, I don't remember how I overcommed the Wx obstacles on OSX, I am
happy an automatic build system works, I just click the "start" button now
and get the OSX builds. Lucky me :-)

> It is a shame, that such a great project like Slic3r is written in such
an inconvenient nerd-on-principle-language.

I am slowly re-writing the computing core from Perl to C++, so the
dependencies to Perl modules will drop gradually. Currently working on the
supports, we will beat Simplify3D! And Slic3r is already bettern than S3D
in some features. For example, S3D does not try hide seams to corners at
all.

MartinMajewski, 2016-11-02:
I have no clue how to tell the building script of Slic3r to look for the cpan Wx install. Perl is such a mess when it comes to versions, locations and dependencies. I think cpan installed Wx somewhere else than cpanm expects it to be and perl is looking for it in the wrong place when running the Slic3r build script. MacOS comes with perl, that is said not to be compatible with Slic3r, so I installed perlbrew a while ago to switch to another perl version. But there is no love lost between macOS perl and perlbrew. As I am not using perl for anything else than the desperate try to build slic3r I don't want to waste time trying to get everything back to the original state. My machine runs perfectly for the Development in C++, D, Swift and Java, so I have no intention to reinstall the system either to get a clean perl installation again to start over messing with perlbrew.

I like slic3r a lot and it is way easier to handle different print-settings in Slic3r than in S3D. That's why I use slic3r more frequently than S3D. But when it comes to adjusting the 3D object (translation, scaling, rotation etc.) and handling difficult parts by manually adjusting the supports, S3D is still better. Not having a convenient object placement interface, no Undo and Redo functionality and no possibility for better support placement in Slic3r is a great pity.

bubnikv, 2016-11-02:
I don't know whether this helps, but you may try to set the environment
variable
PERL_CPANM_HOME
to some of your directories.

Also make sure you are really using the perlbrew perl, by running perl -V,
it will show you all the paths.

Yes, it is a great pain. I am doing it full time for couple of months now,
so I have everything set up, but as a hobby project, it is close to no-go.

lordofhyphens, 2017-02-22:
For 32bit Strawberry perl 5.24.0 and 64bit Strawberry perl 5.18.3, I have archived a copy of local-lib directory at 
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Dependencies/Dependencies-1.3.0-dev

This should have everything necessary to build/run (indeed, it's what I use to avoid rebuilding dependencies on the build server). Make sure you have local::lib installed.

Reference: https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

lordofhyphens, 2017-02-22:
Local::lib usage in current master (and stable!) branches should make this a non-issue.

closed 2017-02-22 by lordofhyphens