slic3r/Slic3r #3489: Feature request: ‘extra length on restart’ proportional to travel distance

DrLex0, 2016-08-30:
I have been browsing through existing issues but found none that match, although #3014 and #1944 come close. I have what might be a stupid idea, but it seems very easy to implement and might help to improve quality for large prints with large extruder travel distances.

I propose to split up the ‘extra length on restart’ field into two: a travel distance scale factor (i.e. X mm extra ‘unretract’ per Y mm travel), and a maximum length. The current behaviour can still be obtained with the maximum as the ‘fixed’ length, and a huge proportional value (or perhaps 0).

The reason why I would like to have this, is that I noticed that my nozzles seem to require the more extra filament to resume printing, the longer they have been sitting idle after a retract. I don't know what causes this, I suspect a combination of multiple factors. Maybe partially due to the loss of oozed material, maybe partially due to lag of pressure changes.

Of course, even if the effect would be perfectly linear with travel time (which it probably isn't), this fix would not be ideal because travel time is not proportional to distance, thanks to acceleration in firmware. But even if it is only approximate, it would be a lot better than the fixed extra length which was only good for squirting ugly blobs everywhere the few times I tried it.

I did some experiments with two cylinders spaced either 16 cm or 8 cm apart, printed external perimeters first, with various extra length settings between 0 and 0.5. It would be a far stretch to say that the experiment shows a perfect linear relation between the size of the gap in the outer perimeter and travel distance. However, using an extra length that is twice the value for the 16 cm distance than for the 8 cm, does produce more consistent results than when using a fixed extra length.

lordofhyphens, 2016-08-30:
You're basically implementing pressure advance in a guesswork fashion.

closed 2016-08-30 by lordofhyphens

DrLex0, 2016-08-30:
Well, it would have been helpful if that checkbox would have a (working) tooltip, because now that I know what it does, I still don't know what to enter, I guess I'll have to dig into the source code. See #2038

lordofhyphens, 2016-08-30:
Hrm, it has a working tooltip in my Linux build of 1.3.0-dev, could you check the development build off of the bintray server (see the README.md for the link) if you're using Windows?

Pressure advance in Slic3r is fairly experimental and was implemented in https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GCode/PressureRegulator.pm and works by trying to model the filament as a spring; the value there is the constant K (which will be different with each filament you use). If you are using Marlin RC7, there's a Linear Advance algorithm implemented in the firmware (which you should use instead of this). I'd start with 1 and do some tests with cylinders or something. The units would be N/m afaik (spring modeled as per Hooke's Law). 

Do not use it with volumetric E or firmware retraction enabled, I've not had good results with either situation. And since all of my systems are configured to use both, well...