slic3r/Slic3r #3490: Feature Request: don`t extdue at Infill turnpoints

Chrisu02, 2016-09-02:
Hy Guy`s haven`t found a open "issue" for this, so i`ll try it:

I analyzed the different slicers more in deph recently and noticed one of the main tricks why cura has such great surfaces.
A reason is that it just extrudes material at the main lines and not on the turnpoints like all of the other slicers do.
Feature Request: don't extrude at turnpoints
[image]

So just skipping the turnpoint lines will make a much smoother surface, it also let the nozzle release pressure in the "missing" gap

As feature i would recommend a optionset to "disable" " the extrusion at the turnpoint (maybe completly or only at the top/bottom layers)

Chri

lordofhyphens, 2016-09-20:
The infill is implemented in perl currently, if you want to take a crack at hacking something together to get what you want. 

If I understand correctly, you are looking for a infill pattern that generates individual strings instead of loops for top solid infill?

Chrisu02, 2016-09-20:
Yes but not only at the top infill but also on the bottom and all inner full infill surfaces (or basicly on all full layers)
This would reduce the "blobs" at the turnpoints especially because most times infill is printed faster than normal perimeters.
This is a known optimization feature cura and meanwhile also craftware do have, but unfortunatly slic3r is still missing this great feature.

lordofhyphens, 2016-09-20:
So you would like all infill to execute in this manner for 100% rectilinear infill? 

(making sure the meaning is clear).

whosawhatsis, 2016-09-21:
Of course, for the length of that movement, without a retraction (which would be terrible, don't do that), plastic will still be coming of of the extruder, with only the back-pressure of the previously extruded plastic to cap-off the nozzle. I would say that you want to go to your travel speed, but for such a short segment, you won't go anywhere near the requested speed anyway.

So you'll get a little extra plastic on the ends, and lose that much on the early part of the following line. This might be why Cura needs the infill/perimeter overlap setting.

Clearly, you don't want to do this for non-solid layers, as I imagine is could significantly weaken the infill.

Chrisu02, 2016-09-21:
@ Lordofhyphens, yes that would only apply to 100% rectilinear infill, sorry forgot to that.
altough Cura is doing this also with the complete (non 100%) infill "Grid" and "Line" patterns, but as whosawhatsis said, it would waken the strengs of the part.

I think G1 would be better because if people use G0 and have a different (higher) acceleration there might be a higher chance of loosing steps ?
and for sure retraction there would be a bad idea ^^

phizz166, 2016-09-21:
I like this idea. I do sometimes notice little peaks and curls at the turning points of solid-infill layers -- that might be addressed by this strategy. (Modifying the extrusion multiplier is not desirable if the rest of the print looks fine).

I'd suggest making a new dropdown in the infill settings specifically for "solid infill," and creating a new pattern option there. Maybe call it "Rectilinear Open" or something.  Alternately, if this is just about the appearance of external surfaces, maybe the new pattern option can just be stuck in the top/bottom infill dropdown.

I'd prefer this to just altering the existing behavior, since you want to retain the ability to fall back if the new strategy isn't working out.

lordofhyphens, 2016-09-21:
Yeah, that would be the plan: add a specific "noloops" or see if the line
pattern works, and make it selectable for other places where 100% infill
would be used.

bubnikv, 2016-09-23:
just set the variable _diagonal_distance of package
Slic3r::Fill::Rectilinear to zero and you will get no connecting lines.

The connecting lines on their own are not the problem. The problem is, that
Slic3r does not account for their volume when calculating infill. This is
not trivial indeed. Current behavior leads quite often to an overfill and
with narrow regions the overfill may be extreme. You may ask for a 20%
infill and get a >100% overfill in some cases.

So disabling the connecting lines is a band aid, not a solution IMHO.

Vojtech

alexrj, 2016-12-17:
I remember many years ago we had that behavior, and many people reported they had inconsistent pressure at the beginning of the lines after the turn. I'm not sure what to do.
If anybody of you wants to experiment, open FillRectilinear.cpp and change this line:

```
if (!this->dont_connect) {
```

to

```
if (!this->dont_connect && this->density < 0.9999f) {
```

and let us know the results :)

lordofhyphens, 2016-12-17:
We could expose it as a config option and not make a gui option for it,
make a wiki page for black magic options. That or keep a separate black
magic config page for these.

Tbh this feels like a lot of "WorksForMe" and using the slicer to
compensate for the machine.

Chrisu02, 2016-12-30:
> I remember many years ago we had that behavior, and many people reported they had inconsistent pressure at the beginning of the lines after the turn. I'm not sure what to do.
If anybody of you wants to experiment, open FillRectilinear.cpp and change this line:
>
> if (!this->dont_connect) {
>
>to
>
>if (!this->dont_connect && this->density < 0.9999f) {
>
>and let us know the results :)

Thanks for that info, unfortunatley is tried now since over a week to compile slic3r master and stable on my Win10 X64 and on a Win7 x86 machine with no success :(
I tried both strawberry and citrusperl, both seem to be compiling the source but when starting slic3r it runs into a _"Can't locate Slic3r/XS.pm in @INC"_

Can maybe somone compile a version for me to a ready exe file so i can check what the resulted gcodes will be ?

thx in advance

lordofhyphens, 2016-12-30:
See README.md.

I am still tweaking the process for builds.

lordofhyphens, 2016-12-30:
Note: the "exe" is an artifact of the Perl packager @alexrj used (Cava
Packager). The auto builds use PAR:: Packer and instead provide a batch
file to make things more convenient.

lordofhyphens, 2016-12-31:
@Chrisu02 https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-1.3.0-dev.2016.12.31.2be6e1a.zip *should* work, although you might need to use the CLI interface. I've been having some issues with builds off of the appveyor server, but this ran in a separate directory on my laptop.

Chrisu02, 2016-12-31:
Thx for your effort @ lordofhyphens  :)
I tried out your version, it will start, but the infill has still all turnpoints/lines ?
I have checked the folders, your version doesn`t have a "/lib/Slic3r/Fill" folder anymore an i did not found the FillRectilinear.cpp also ?
so i don`t know where i should make the changes now.

lordofhyphens, 2016-12-31:
Infill is now in c++. This means you will have to get it to build to make
changes. Try the cubic or triangle infill btw.

lordofhyphens, 2017-02-08:
@Chrisu02 
Does the new rectilinear infill at https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=slic3r-master.2017.01.26.348.d41e3c1.zip serve your purpose?

It was written by @bubnikv as Rectilinear2 and then @alexrj revised it and made it the regular rectilinear fill.

bubnikv, 2017-02-08:
What is asked for here is to set the variable link_max_length to FLT_MAX, so the infill lines will never be connected along the perimeters, but to do it for the top / bottom surfaces only, or to make the variable configurable through the UI, so for the top / bottom surfaces long links will not be drawn along the perimeters. It certainly makes sense to try it.

Chrisu02, 2017-02-08:
Thanks for your efford @ lordofhyphens

i just tired out this version, for what i can see you have exchanged the line infill to "Aligned rectilinear", thats right ?
I don`t see a "rectilinear2" there ?

i just sliced an object, but i can`t see any difference between normal rectilinear and alligned rectilinear.
on both infills the turnpoints(lines) are still filled with material ?

ah and please don`T remove the line infill on further versions, thats the normal infill i use ^^

Chri

lordofhyphens, 2017-02-08:
Infill has been completely rewritten, Line is going to be gone until
someone gets around to implementing it again. In the meantime, there is
Triangle fill (IIRC) or cubic (which is a 3d infill).

Check to make sure if Rectilinear (it is new) works for you.

lordofhyphens, 2017-02-08:
My point is that it probably won't overfill now.

Chrisu02, 2017-02-08:
Oh ok, then i`ll try it out at the weekend.

Chrisu02, 2017-02-11:
Hy Lordofhyphens, i just made two quick test parts, saw no difference.
i also analyzed the gcode, there is no real difference there.,

> Infill has been completely rewritten, Line is going to be gone until someone gets around to implementing it again. In the meantime, there is Triangle fill (IIRC) or cubic (which is a 3d infill).

Maybe my text war written in a missunderstanding way, but the main reason for not fill at turnpoints is not for the infill inside the part (non 100% fill) but for the Bottom and Top layers.
at the inside infill the "overextruded" material has space to run inside the part, but at the 100% bottom/top layers there is no room for that.

Chri