slic3r/Slic3r #3491: Entire layer is skipped (empty)

DrLex0, 2016-09-03:
### Version
1.2.9
Also tested it in the 1.30.0.25 prusa3d fork, same behavior.

### Operating system type + version
Mac OS X 10.11.6

### Behavior
Slice the attached STL file (which is manifold) with attached settings.
The entire layer at Z position 4.10 is missing.
My curious self would like to see what happens if I print this, but my sane self tells me it will be a waste of time and perfectly good ABS.

Also attached is a modified STL that no longer exhibits the problem, the only thing I changed is dissolve some of the edges in the surface at the 4.10 mm Z position.

#### STL/Config (.ZIP) where problem occurs
[IssueFiles.zip](https://github.com/alexrj/Slic3r/files/453471/IssueFiles.zip)

lordofhyphens, 2016-09-03:
I suspect that if you had a terminal window open for debug message it'd be complaining about not being able to complete loops.

lordofhyphens, 2016-09-03:
>>  Unable to close this loop having 20 points
Something is definitely up with the handling of the file, and whatever it is isn't checked during the manifold checking. 
[image]

According to Slic3r's preview, it is making a double-thick layer at 4.3.0, and it's weird in other ways.

[image]

DrLex0, 2016-09-03:
It would be nice if slicing errors, or even just a warning icon, would show up somewhere in the interface, so users could know it wouldn't be a good idea to just start printing.

I usually remove redundant edges resulting from boolean operations, but this time I had been lazy. Apparently it is good hygiene after all…

lordofhyphens, 2016-09-20:
Errors like that show up in the console (stdout), probably because that's coming from libslic3r as a printf statement and isn't redirected to somewhere the GUI can understand it. 

Probably worth a refactor.

bubnikv, 2016-09-23:
It is worth looking into the Cura code for slicing the STL. I have a
feeling its automatic repair works better, or maybe it will just give the
user a chance to try an alternate strategy.