slic3r/Slic3r #3495: Automatically fill pattern does not change

AndylgTom, 2016-09-07:
### Version
compile the repository 1.30 (date 05.09.2016)....

### Operating system type + version
Windows10 x64

### Behavior
Automatically fill pattern does not change, when choosing a 100% fill density.
In version 1.29 and in the Assembly of the Prusa, everything is fine.

GIF
[image]

lordofhyphens, 2016-09-07:
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches should have a copy of the branch built for win32 in about an hour or so (from posting). Check the related pull request for the branch name.

AndylgTom, 2016-09-20:
Downloaded an updated dev version of 19.09. Nothing has changed. After compilation, the problem remained.

You probably do not understand what is the problem?
It must be like this gif:
[image]

lordofhyphens, 2016-09-20:
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches/1.3.0-fix-3495-infill-100-ui-185

I found the problem you posted about originally. Above is a link with a fix.

lordofhyphens, 2016-09-20:
https://github.com/alexrj/Slic3r/pull/3496/files is a copy of the fix itself.

lordofhyphens, 2016-09-20:
I can't reproduce problem on Linux in first post with my fix in place,
behavior looks like second GIF.

AndylgTom, 2016-09-20:
Windows 10x64  :( Strange :(

lordofhyphens, 2016-09-20:
This is the expected behavior for that dialog box:
[image]

Honeycomb infill is not permitted for 100% infill.

AndylgTom, 2016-09-20:
Of course, fill pattern must change to Rectilinear. This is not happening, when click the Ok button (like in the first gif)

bubnikv, 2016-09-23:
May this bug be caused by an outdated GUI library on Windows @alexrj and
@LoH are using? I am happy to hear it works with Prusa3D builds :-)

lordofhyphens, 2017-03-26:
Cannot replicate with copy from build server. @alexrj fixed this in https://github.com/alexrj/Slic3r/commit/376df2e3f2335c221156515de2bd48cedc7bb430

lordofhyphens, 2017-03-26:
https://bintray.com/version/files/lordofhyphens/Slic3r/slic3r_dev/1.3.0-dev-2fc8152 tested build

closed 2017-03-26 by lordofhyphens