slic3r/Slic3r #3497: Versions

Tinchus2009, 2016-09-07:
### Version
_Version of Slic3r used goes here_

This is actually my problem: the versions. I installed 1.3 dev version a cpuple of mounths ago. Tody I wanted to update, so I did the usual git pull command, perl build.pl commands.
Everything went ok. I want to report a couple of bugs I noticed, so I did a git describe --tag. 

What I get is 1.2.9-364-g3700950 ..... 
If I open slic3r , in the HELP > ABOUT SLIC3R it says version 1.3.0 DEV
What is happening here? how do I stay updated on the latest corrections?, definitly 1.3 DEV works much better tha the "stable" 1.2.9

lordofhyphens, 2016-09-07:
Git describe looks at the last tag for the version. @alexrj hasn't tagged
v1.3.0 so it would not say that in git.

The best way to know what your are using if it is from git is to look at
the commit hash. If it isn't the most recent commit, it's old ;)

lordofhyphens, 2016-09-07:
If you want to report bugs from -git, run git log and copy the topmost hash
value and github will link it.

closed 2016-09-07 by lordofhyphens